MarlinFirmware/Marlin #10737: Bed leveling on y axis is backwards.

TheMagicDick, 2018-05-14:
Printer: monoprice maker select v2 with rambo einsy board
marlin version: 1.1.5 einsy fork

So when using the bed visualizer plugin with octoprint it shows that the mesh has flipped y values. I noticed i had leveling issues before and this just confirmed it. I would raise 0,0 of my bed and sure enough the bed visualizer showed that it was raised but with the y value of 200. I can print normally if i manually level the bed (which defeats the purpose of the z probe) and the y axis isn't flipped when printing when i manually level the bed. 

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1999083/Configuration.zip)

Squid116, 2018-05-14:
99.99% sure this is the way the plugin interprets the results of G29 T.  I too see the reversal, but I am 100% sure my bed is working and not flipped - there is a fair bow on one side only, and I'd notice if it was flipped...

TheMagicDick, 2018-05-15:
I will further try to confirm that it's not just the plugin. And I will update to the new bug fix. I won't be able to touch my printer for 2-3 days. But thank you for the replies

TheMagicDick, 2018-05-15:
I have confirmed that squid is right. The real problem isn't that it's backwards it's that it doesn't seem to make a correct amount of z corrections. I can feel the z axis rise or fall at the correct time but it doesn't do enough corrections to keep a level bed.

Squid116, 2018-05-15:
Have you got software end mstops enabled? I did for ages, and this stops it going below z=0 so it can't lower to the low points of the bed.

TheMagicDick, 2018-05-16:
I have only the max software endstop enabled, min software endstop is
disabled. I think i have the probe offset a bit off. when i home the z
probe it isnt perfectly in the center of the bed, its close, like 10 mms
off to the right and 5 mm off in the y direction. Do you think that would
make that much a difference?

Squid116, 2018-05-16:
I believe small errors in the probe offset produces rather unpredictable results in bed levelling.

TheMagicDick, 2018-05-16:
I watched a toms video and he said it just had to be in the ball park but i
will experiment and see if that would fix it. Thank you for reply.

thinkyhead, 2018-05-16:
> Have you got software end mstops enabled? I did for ages, and this stops it going below z=0 so it can't lower to the low points of the bed.

Software endstops don't affect movement below 0 related to bed leveling compensation. They are applied to the logical coordinates at the G-code level.

TheMagicDick, 2018-06-07:
I'm sorry for posting so late have not been able to mess around with my printer do some vacationing. But i did redo the x probe offset to be more accurate and still see that the leveling issue has not been corrected. I am now thinking that it may be due to my bed being warped. It always seems that the closer it gets to the top left of the bed the more it bottems out. any suggestions before i buy an aluminum build plate. 

P.S. Next chance i get i will turn off the leveling system and just do normal leveling via the paper test on each corner and see if i can tell if it is warped or not.

thinkyhead, 2018-06-07:
A twisted X gantry can also lead to incorrect probe readings, high on one side and low on the other. Check to make sure the carriage stays perfectly vertical as it moves.

TheMagicDick, 2018-06-16:
What do you recommend is the best way of doing this. I've printed levelers
that attach to the y axis rails to help level the x axis. I've also tried
making the x axis go all the way up and hit the frame at the top to
hopefully level it.

thinkyhead, 2018-06-22:
If the X gantry is simply _**leaning**_, then auto bed leveling will compensate for that. But if it is _**twisted**_ (therefore causing the probe to change its front-to-back angle depending on its left-to-right position) then you need to do other kinds of physical adjustments. For example, adjust the belt tension, tighten or loosen the wheels that ride on the X gantry, etc.

TheMagicDick, 2018-06-24:
> Ok I understand what your saying now. The belt is tight enough I'll try to
> make sure every thing is straight and squared off. I'll get back to you.

Also I have a modded z brace on my printer. Do you think that can cause
this problems If I set it up wrong? I'm gonna check the next time I can,
but do you think that this could be the problem.

thinkyhead, 2018-06-30:
Any number of factors might lead to unreliable probe readings. When readings are valid we know the bed leveling system works. While troubleshooting your probe setup, you can switch to one of the manual bed leveling systems in the meantime (i.e., `MESH_BED_LEVELING` or bilinear with `PROBE_MANUALLY`) so that you can at least have a good mesh saved to EEPROM for daily use.

TheMagicDick, 2018-06-30:
I finally found the problem i believe. I haven't been about to extensively
test it (due to one of my wires for my extruder motor broke) but from what
i tested it is give a nice level first layer. As i suspected from my last
post it was my z brace. I made sure i was more accurate when adjusting the
brace to make sure that the z axis frame was perpendicular to the desk.
Thank you for helping i believe this is solved.

closed 2018-06-30 by thinkyhead