MarlinFirmware/Marlin #10738: [bugfix-1.1.x] UBL causes skips when crossing the calibrated area

orcinus, 2018-05-14:
### Description

If calibrated area is smaller than the bed / usable area (which it has to be in my case, because with a dual independent X carriages, i get crashes into the second carriage during calibration), UBL causes a visible and audible skip when moving across the border of the calibrated area (i.e. for Xmax-inset < X < X0+inset, and Ymax-inset < Y < Y0+inset).

I'm calibrating manually.
Do i need to do something (interpolation?) to avoid this?

### Steps to Reproduce

1. Enable UBL
2. Configure an inset for UBL
3. Level the bed manually
4. Try a continuous move beyond the inset border on either X or Y

**Expected behavior:**

Smooth move.

**Actual behavior:**

Planner breaks it up into two moves, and creates a noticeable pause inbetween.

Roxy-3D, 2018-05-14:
I'll try to remember to try this tomorrow.   I have a printer with a large inset.   I should be able to duplicate what you are seeing on that printer.    If I don't comment, it means I forgot.   And if that happens... Please remind me this is still open.

thinkyhead, 2018-05-14:
@Roxy-3D — Is it true that because of the way UBL is implemented, there may be a large and sudden change in Z between calibrated areas and non-calibrated areas? Or should it smoothly interpolate, just as if the non-calibrated points were zeros?

Bergerac56, 2018-05-14:
I fully confirm that. I already mentionned this long time ago (Summer last year). When out of the mesh, Z jumps. It would be greath to have an interpolation in those cases.

Roxy-3D, 2018-05-14:
> Is it true that because of the way UBL is implemented, there may be a large and sudden change in Z between calibrated areas and non-calibrated areas? Or should it smoothly interpolate, just as if the non-calibrated points were zeros?

I guess it depends on what you mean by 'uncalibrated'.    If the mesh is not fully built, the mesh point is a NAN and the Z-Height correction number ends up being 0.    So, that can cause a jump.   But going off the mesh is also 'uncalibrated'.   Currently that is handled by allowing the user to specify a Z-Height to use.    This needs to be improved upon.   Extrapolating the nearest known mesh value might be one way to do that.   Extrapolating the contour of the mesh at the nearest location is probably a better (and more expensive) way of doing that.

orcinus, 2018-05-14:
> Extrapolating the nearest known mesh value might be one way to do that. Extrapolating the contour of the mesh at the nearest location is probably a better (and more expensive) way of doing that.

How about an even simpler / lighter solution - just extending the last known Z correction value for a particular X and Y in the direction of the axis? This leaves the corners, which get a simple single-value fill from the corner of the measured area. Kinda like this:

https://i.imgur.com/c9JImI9.png

If that makes sense. Not very accurate, but it doesn't have to be - it just needs to be a seamless transition. Should be fairly light-weight processingwise too.

Roxy-3D, 2018-05-14:
What is your opinion about having invalid (NAN) numbers in the bed's mesh?   Should auto bed leveling still be available?   Should the user be able to turn it on?

I can add an option to allow extrapolation off the mesh.   And possibly do that in multiple flavors.  But I'm worried about damaging somebody's machine.

orcinus, 2018-05-14:
I don't think a matrix with NAN elements should be considered valid / should allow bed leveling to work. Unless replaced with interpolated values on the fly, of course.

> I'm worried about damaging somebody's machine.

Frankly, i'm more worried about someone not having the fallback Z value for undefined areas of the mesh set up properly, and having their nozzle suddenly drop into a print or something. I don't know what kind of beds people run, but i get very small variations in the mesh, and it's mostly due to the droop of my X and Y axes, not the actual bed unevenness (speaking of which - i wonder what happens when you have Z Fade on, and most of your mesh variation comes from axis droop, not the bed).

orcinus, 2018-05-14:
Unrelated (sorry for reusing this issue), but is there any way i can solve my original issue - which is UBL with a dual independent X carriage?

I added a hefty inset because, while my bed's X is 300mm, the first carriage can't physically get to X=300 without hitting the second X carriage. I can't define bed size as <300mm because that doesn't make sense, as *the second* X carriage *can* get to X=300.

Adjusting UBL_MESH_MAX_X to <300 is the same as having an inset (i.e. i end up with the second carriage outside the mesh).

thinkyhead, 2018-05-14:
> I don't think a matrix with NAN elements should be considered valid / should allow bed leveling to work.

If we add a `ubl.mesh_is_valid()` method and call it from `leveling_is_valid` that would be enough to prevent UBL being enabled when not filled-in.

Roxy-3D, 2018-05-14:
OK.

boelle, 2019-02-20:
@orcinus has the issue gone away in latest bugfix 2.0?

orcinus, 2019-02-23:
To be completely honest, i've been staying away from 2.0 and have reverted to using a 1+ year old build for the past several months, as the new versions were simply to unreliable for serious use.

I'll give it a shot over the weekend and close if it's solved.

boelle, 2019-03-06:
@orcinus did you test with 2.0?

@thinkyhead i think we can close this one

orcinus, 2019-03-06:
Never got around to testing it, unfortunately, because i had a working weekend.
Feel free to close, i can always reopen as 2.0 issue if needed, when i get around to testing it.

boelle, 2019-03-06:
@orcinus only you can close, button is below

orcinus, 2019-03-06:
Sorry, i'm daft, been commenting on other people's issues whole day at work. Closing.

closed 2019-03-06 by orcinus