MarlinFirmware/Marlin #10739: steval-3dp001v1 support?

ffnull, 2018-05-14:
Can someone help with compiling brunch 2.0 firmware for steval-3dp001v1?

thinkyhead, 2018-05-14:
Is that STM32F1? Follow this guide and see how far you can get:
http://marlinfw.org/docs/basics/install_platformio.html

xC0000005, 2018-05-15:
That's an STM32F4 board, so you'll need to install STM32GENERIC - see the readme.md in the STM32F4 HAL folder.

ffnull, 2018-05-15:
Im try installing STM32GENERIC on linux but its crashd. Can you explain in short how to ?

xC0000005, 2018-05-16:
The arduino IDE is crashing? Or PlatformIO is crashing? Or you're getting an exception building? I can probably help but I need to know what is happening to help.

ffnull, 2018-05-16:
I'm working on arch Linux with platformio. I'm can't install stm32generic. To many errors in platformio. Can't find normal instruction to install.im trying this metod http://www.stm32duino.com/viewtopic.php?t=2066

ffnull, 2018-05-16:
`Error: 
[Thu May 17 03:44:59 2018] Processing STM32F4 (platform: https://github.com/platformio/platform-ststm32.git; board: disco_f401vc; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: ST STM32 > ST 32F401CDISCOVERY
SYSTEM: STM32F401VCT6 84MHz 64KB RAM (256KB Flash)
DEBUG: CURRENT(stlink) ON-BOARD(stlink) EXTERNAL(blackmagic, jlink)
Converting Marlin.ino
Error: This board doesn't support arduino framework!
========================== [ERROR] Took 0.35 seconds ==========================
================================== [SUMMARY] ==================================
Environment megaatmega2560          	[SKIP]
Environment megaatmega1280          	[SKIP]
Environment DUE                     	[SKIP]
Environment DUE_USB                 	[SKIP]
Environment DUE_debug               	[SKIP]
Environment LPC1768                 	[SKIP]
Environment LPC1768_debug_and_upload	[SKIP]
Environment melzi                   	[SKIP]
Environment melzi_optiboot          	[SKIP]
Environment rambo                   	[SKIP]
Environment sanguino_atmega644p     	[SKIP]
Environment sanguino_atmega1284p    	[SKIP]
Environment STM32F1                 	[SKIP]
Environment STM32F4                 	[ERROR]
Environment teensy20                	[SKIP]
Environment teensy35                	[SKIP]
Environment malyanm200              	[SKIP]
========================== [ERROR] Took 0.35 seconds ==========================`

`[env:STM32F4]
platform    = https://github.com/platformio/platform-ststm32.git
framework   = arduino
board       = disco_f401vc
build_flags = -DUSE_STM32GENERIC -DMENU_USB_SERIAL
lib_deps    = ${common.lib_deps}
lib_ignore  = Adafruit NeoPixel, c1921b4, TMC2130Stepper
src_filter  = ${common.default_src_filter}
monitor_speed = 250000`

xC0000005, 2018-05-16:
Check the board’s .json file in the platformio/platforms/ststm32/boards (it’s dsco_f407vg.json) - odds are the framework “arduino” isn’t listed in that board’s frameworks list. go ahead and add it. I didn’t change every board in the repo because I am not certain all of them work.

ffnull, 2018-05-16:
return self.method(*nargs, **kwargs)
File "/home/ffnull/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/platformio.py", line 69:
env.BuildFrameworks(env.get("PIOFRAMEWORK"))
File "/home/ffnull/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/home/ffnull/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/platformio.py", line 278:
SConscript(env.GetFrameworkScript(f))

File "/home/ffnull/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 604:
return method(*args, **kw)
File "/home/ffnull/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "/home/ffnull/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "/home/ffnull/.platformio/platforms/ststm32/builder/frameworks/arduino/stm32duino.py", line 29:
if "stm32f1" in env.BoardConfig().get("build.variant"):
File "/home/ffnull/.platformio/penv/lib/python2.7/site-packages/platformio/managers/platform.py", line 666:
raise KeyError("Invalid board option '%s'" % path)

ffnull, 2018-05-16:
This error after im adding arduino to disco_f401vc.json
`
{
  "build": {
    "cpu": "cortex-m4",
    "extra_flags": "-DSTM32F401xC",
    "f_cpu": "84000000L",
    "mcu": "stm32f401vct6"
  },
  "frameworks": [
    "arduino",
    "stm32cube"
  ],
  "name": "ST 32F401CDISCOVERY",
  "upload": {
    "maximum_ram_size": 65536,
    "maximum_size": 262144,
    "protocol": "stlink"
  },
  "debug": {
    "tools": {
      "stlink-v2": {
        "server": {
          "package": "tool-openocd",
          "executable": "bin/openocd",
          "arguments": ["-f", "scripts/board/stm32f4discovery.cfg"]
        },
        "onboard": true
      }
    }
  },
  "url": "http://www.st.com/web/catalog/tools/FM116/SC959/SS1532/LN1848/PF259098",
  "vendor": "ST"
}
`

xC0000005, 2018-05-16:
If you look at your platformio/platforms/ststm32/builder/frameworks/arduino folder, do you have an STM32GENERIC folder there with a builder .py file in it?

ffnull, 2018-05-17:
cant find builder.py Im doing something wrong. Can you provide step by step instruction by installing?

ffnull, 2018-05-17:
????

xC0000005, 2018-05-17:
I’ll write it up when I get a chance to get a clean machine, but I think the key here is you’re going to (at a command line) run:
pio platform uninstall ststm32
pio platform install https://github.com/xC0000005/platform-ststm32#M200Toolchain <https://github.com/xC0000005/platform-ststm32#M200Toolchain>

I haven’t had a chance to install on a clean machine from scratch, but that should get you my updated builder. You’d then need to add in arduino as a framework again, then 

Do this:
cd ~\.platformio\packages
git clone https://github.com/xC0000005/STM32GENERIC#CoreUpdates
mv STM32GENERIC framework-arduinoSTM32GENERIC

Note I already added a platform.json, so that shouldn’t be needed now.
I’ll be taking a clean machine and installing atom from scratch soon to test this end to end.

xC0000005, 2018-05-19:
I tested - using the PIO install method I documented above (the #<branch> name) gets you my builder and so on. I'm trying to figure out how to add a manifest so the STM32Generic just installs as well, but that means finding a place to host the tarball for each release. Github rejects my attempt to push it there, and using the /archive/ method works, but breaks the hash if I make a commit. Not sure how I'm going to resolve that yet.

ffnull, 2018-05-19:
What im try.
1)Clean install of platformio
2)Open project marlin bugfix 2.0
3)Edit platformio.ini
`
[platformio]
src_dir = Marlin
build_dir = .pioenvs
lib_dir = .piolib
libdeps_dir = .piolibdeps
env_default = STM32F4

[common]
default_src_filter = +<src/*> -<src/config>
build_flags = -fmax-errors=5
  -g
  -ggdb
lib_deps =
  https://github.com/MarlinFirmware/U8glib-HAL/archive/dev.zip
  LiquidCrystal_I2C@1.1.2
  TMC2130Stepper
  https://github.com/teemuatlut/TMC2208Stepper/archive/v0.1.1.zip
  Adafruit NeoPixel@1.1.3
  https://github.com/lincomatic/LiquidTWI2/archive/30aa480.zip
  https://github.com/ameyer/Arduino-L6470/archive/master.zip
  https://github.com/trinamic/TMC26XStepper/archive/c1921b4.zip

[env:STM32F4]
platform      = ststm32
framework     = arduino
board         = disco_f401vc
build_flags   = ${common.build_flags} -DUSE_STM32GENERIC -DMENU_USB_SERIAL -DMENU_SERIAL=SerialUSB
lib_deps      = ${common.lib_deps}
lib_ignore    = Adafruit NeoPixel, c1921b4, TMC2130Stepper
src_filter    = ${common.default_src_filter}
monitor_speed = 250000
`
4)pio platform uninstall ststm32
5)pio platform install https://github.com/xC0000005/platform-ststm32#M200Toolchain
6)cd ~\.platformio\packages
7)git clone -b CoreUpdates https://github.com/xC0000005/STM32GENERIC
8)mv STM32GENERIC framework-arduinoSTM32GENERIC
9)Go to .platformio\platforms\ststm32\boards\disco_f401vc.json and addind arduino
`
{
  "build": {
    "cpu": "cortex-m4",
    "extra_flags": "-DSTM32F401xC",
    "f_cpu": "84000000L",
    "mcu": "stm32f401vct6"
  },
  "debug": {
    "default_tools": [
      "stlink"
    ],
    "onboard_tools": [
      "stlink"
    ],
    "openocd_board": "stm32f4discovery",
    "openocd_target": "stm32f4x",
    "svd_path": "STM32F401x.svd"
  },
  "frameworks": [
    "arduino",
    "stm32cube"
  ],
  "name": "ST 32F401CDISCOVERY",
  "upload": {
    "maximum_ram_size": 65536,
    "maximum_size": 262144,
    "protocol": "stlink",
    "protocols": [
      "jlink",
      "stlink",
      "blackmagic"
    ]
  },
  "url": "http://www.st.com/web/catalog/tools/FM116/SC959/SS1532/LN1848/PF259098",
  "vendor": "ST"
}
`
10)After all im get error:
`
[05/20/18 02:34:39] Processing STM32F4 (platform: ststm32; board: disco_f401vc; framework: arduino)
 
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: ST STM32 > ST 32F401CDISCOVERY
SYSTEM: STM32F401VCT6 84MHz 64KB RAM (256KB Flash)
DEBUG: CURRENT(stlink) ON-BOARD(stlink) EXTERNAL(blackmagic, jlink)
Converting Marlin.ino

stm32generic()
disco_f401vc --> DISCOVERY_F401VC
KeyError: "Invalid board option 'build.variant'":
File "C:\users\ffnull\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 163:
env.SConscript("$BUILD_SCRIPT")
File "C:\Users\ffnull\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "C:\Users\ffnull\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "C:\users\ffnull\.platformio\platforms\ststm32\builder\main.py", line 127:
target_elf = env.BuildProgram()
File "C:\Users\ffnull\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\users\ffnull\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 69:
env.BuildFrameworks(env.get("PIOFRAMEWORK"))
File "C:\Users\ffnull\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\users\ffnull\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 278:
SConscript(env.GetFrameworkScript(f))
File "C:\Users\ffnull\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 604:
return method(*args, **kw)
File "C:\Users\ffnull\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "C:\Users\ffnull\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "C:\users\ffnull\.platformio\platforms\ststm32\builder\frameworks\arduino\stm32duino.py", line 31:
if "stm32f1" in env.BoardConfig().get("build.variant"):
File "c:\users\ffnull\.platformio\penv\lib\site-packages\platformio\managers\platform.py", line 666:
raise KeyError("Invalid board option '%s'" % path)
 [ERROR] Took 1.01 seconds
`

xC0000005, 2018-05-20:
This is progress - see the stm32Generic()? That means you are getting closer - it’s invoking the right build chain. Go ahead and treat this sucker like it’s a discovery, and set your board to a disco_f407vg, which actually has a proper variant. Long term we’re going to need to add one for your board, but you can get compiling.
You can see what the core supports by looking at .platformio/packages/package-arduinoSTM32GENERIC/STM32/VARIANTS - the builder maps anything that starts with “disco_” to one of the discovery variants, so let’s pick one it actually has. (alternately, you can add the “variant” property to the build section of the board.json file you picked. 

But I’d choose the path of least resistance. Once you can actually get a binary to compile you can get fancy and switch to your own variant and all that jazz.

ffnull, 2018-05-20:
Still same error with disco_f407vg

xC0000005, 2018-05-20:
Ok. Let me see what I can do to get a clean machine setup and test.

xC0000005, 2018-05-21:
Sorry, my daughter was in town for a week-long visit, and this was much lower priority. 

I’ve been going over the state of STM32 cores, and it appears there are more than I ever thought. The STM32GENERIC core which F4 and F7 HALS are built on doesn’t appear to heavily maintained. There’s a fork for a user named huaweiwx which appears to be maintained (but doesn’t support the board that the F7 support was added for). There’s also this one - https://github.com/stm32duino/Arduino_Core_STM32 <https://github.com/stm32duino/Arduino_Core_STM32>, which appears to be updated constantly but doesn’t work in PIO. 

Here’s a thing you could try that would be cheap and easy to try so you can get unblocked and move on to fun things like pins. The STM32F1 HAL actually supports F4 as well (it came first). If you add to your build flags in platformio.ini  -D__STM32F1__ -DTARGET_STM32F1, it will build using that HAL for STM32F4

You’d set your board to something like genericSTM32F407VET6 and go ahead and give it a whirl. This should compile using the original STM32 HAL (named “STM32F1”) but with support for F4. @victorpv was building for STM32F4 before the current F7/F4 hal was incorporated doing this.

At some point there needs to be a reckoning over how all this is handled, but that’s a problem for when it’s easy to build across multiple boards.

For your pins definitions, btw, the marlin source for that eval board uses ints, but all you have to do is open the macro map and read the port + pin number for  the index.

ffnull, 2018-05-21:
Nope. Maybe im total idiot. But im have same error. Can you prowide short step by step to check what im missing?

xC0000005, 2018-05-21:
Maybe in your platformio section, your build flags still containse -DUSE_STM32GENERIC - that switch is what chooses if you build with the maple stuff or STM32generic.

> On May 21, 2018, at 2:03 PM, ffnull <notifications@github.com> wrote:
> 
> step

xC0000005, 2018-05-22:
I think I figured out what is causing you problems with the build.variant not defined, and I pushed a commit to the M200 toolchain branch. You can reinstall the package, or modify the stm32duino.py file under your PIO builder:
if "USE_STM32GENERIC" in env.get("CPPDEFINES"):
    env.SConscript("STM32GENERIC/stm32generic.py")
if "stm32f1" in env.BoardConfig().get("build.variant"):

should be:
if "USE_STM32GENERIC" in env.get("CPPDEFINES"):
    env.SConscript("STM32GENERIC/stm32generic.py")
elif "stm32f1" in env.BoardConfig().get("build.variant"):

The way this works is that if your board defines have USE_STM32GENERIC in them, it should detour to the generic builder, but it was also running the normal ones.

ffnull, 2018-05-23:
ok. im got new error

TypeError: object of type 'NoneType' has no len():
File "C:\users\ffnull\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 163:
env.SConscript("$BUILD_SCRIPT")

File "C:\Users\ffnull\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "C:\Users\ffnull\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "C:\users\ffnull\.platformio\platforms\ststm32\builder\main.py", line 127:
target_elf = env.BuildProgram()
File "C:\Users\ffnull\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\users\ffnull\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 69:
env.BuildFrameworks(env.get("PIOFRAMEWORK"))
File "C:\Users\ffnull\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\users\ffnull\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 278:
SConscript(env.GetFrameworkScript(f))
File "C:\Users\ffnull\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 604:
return method(*args, **kw)
File "C:\Users\ffnull\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "C:\Users\ffnull\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "C:\users\ffnull\.platformio\platforms\ststm32\builder\frameworks\arduino\stm32duino.py", line 30:
env.SConscript("STM32GENERIC/stm32generic.py")
File "C:\Users\ffnull\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "C:\Users\ffnull\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "C:\users\ffnull\.platformio\platforms\ststm32\builder\frameworks\arduino\STM32GENERIC\stm32generic.py", line 229:
stm32generic()
File "C:\users\ffnull\.platformio\platforms\ststm32\builder\frameworks\arduino\STM32GENERIC\stm32generic.py", line 74:
"framework-arduinostm32generic"), "STM32")
File "c:\users\ffnull\.platformio\penv\lib\ntpath.py", line 65:
result_drive, result_path = splitdrive(path)
File "c:\users\ffnull\.platformio\penv\lib\ntpath.py", line 115:
if len(p) > 1:

xC0000005, 2018-05-24:
Attach for me your platformio.ini and I'll pull it down and compile. If it fails for me, I can figure out what you need changed and adjust it. If it doesn't, then we'll know my builder and yours are not identical, and we can fix that fairly easily.

ffnull, 2018-05-25:
#
# Marlin Firmware
# PlatformIO Configuration File
#
# For detailed documentation with EXAMPLES:
#
# http://docs.platformio.org/en/latest/projectconf.html
#

# Automatic targets - enable auto-uploading
# targets = upload

#
# By default platformio build will abort after 5 errors.
# Remove '-fmax-errors=5' from build_flags below to see all.
#

[platformio]
src_dir = Marlin
build_dir = .pioenvs
lib_dir = .piolib
libdeps_dir = .piolibdeps
env_default = STM32F4

[common]
default_src_filter = +<src/*> -<src/config>
build_flags = -fmax-errors=5
  -g
  -ggdb
lib_deps =
  https://github.com/MarlinFirmware/U8glib-HAL/archive/dev.zip
  LiquidCrystal_I2C@1.1.2
  TMC2130Stepper
  https://github.com/teemuatlut/TMC2208Stepper/archive/v0.1.1.zip
  Adafruit NeoPixel@1.1.3
  https://github.com/lincomatic/LiquidTWI2/archive/30aa480.zip
  https://github.com/ameyer/Arduino-L6470/archive/master.zip
  https://github.com/trinamic/TMC26XStepper/archive/c1921b4.zip

#################################
#                               #
#   Unique Core Architectures   #
#                               #
#  Add a new "env" below if no  #
# entry has values suitable to  #
#   build for a given board.    #
#                               #
#################################

#
# ATmega2560
#
[env:megaatmega2560]
platform      = atmelavr
framework     = arduino
board         = megaatmega2560
build_flags   = ${common.build_flags}
board_f_cpu   = 16000000L
lib_deps      = ${common.lib_deps}
src_filter    = ${common.default_src_filter}
monitor_speed = 250000

#
# ATmega1280
#
[env:megaatmega1280]
platform      = atmelavr
framework     = arduino
board         = megaatmega1280
build_flags   = ${common.build_flags}
[... 245 lines cut ...]
lib_deps      = ${common.lib_deps}
lib_ignore    = Adafruit NeoPixel
src_filter    = ${common.default_src_filter}
monitor_speed = 250000

[env:malyanm200]
platform    = ststm32
framework   = arduino
board = malyanM200
build_flags = !python Marlin/src/HAL/HAL_STM32F1/stm32f1_flag_script.py -DMCU_STM32F103CB -D __STM32F1__=1 -std=c++1y -D MOTHERBOARD="BOARD_MALYAN_M200" -DSERIAL_USB -ffunction-sections -fdata-sections -Wl,--gc-sections
src_filter = ${common.default_src_filter}
#-<frameworks>
lib_ignore =
  U8glib
  LiquidCrystal_I2C
  LiquidCrystal
  NewliquidCrystal
  LiquidTWI2
  Adafruit NeoPixel
  TMC2130Stepper
  Servo(STM32F1)
  TMC26XStepper
  U8glib-HAL
  TMC2208Stepper
  c1921b4

xC0000005, 2018-05-25:
Ok. I went ahead and created a bare-bones variant for you based off the right CPU and variant, and added a section specifically for your board.
Try removing the framework-arduinoSTM32GENERIC from your .platformio/packages directory, 
then adding this to your platformio.ini:

[env:steval3dp]
platform      = https://github.com/xC0000005/platform-ststm32.git#M200Toolchain
framework     = arduinostm32generic
board         = stm323dpeval
build_flags   = ${common.build_flags} -DMENU_USB_SERIAL -DMENU_SERIAL=SerialUSB -DUSE_USB -D MOTHERBOARD="BOARD_STM32F4" -DSTM32F4 -DSTM32Fx -DUSE_USBSERIAL
lib_deps      = ${common.lib_deps}
lib_compat_mode = off
lib_ignore    = Adafruit NeoPixel, c1921b4, TMC2130Stepper, TMC26XStepper
src_filter    = ${common.default_src_filter}
monitor_speed = 250000

and running with with pio run -v -e steval3dp

I get a successful compile, but I can't say it works, just that it's a start.

ffnull, 2018-06-02:
Yes. Its starts compile. But right now problemm with hal

xC0000005, 2018-06-02:
What problems?

ffnull, 2018-06-05:
```
*** [.pioenvs\steval3dp\src\src\HAL\HAL_STM32F4\HAL_spi_STM32F4.o] Error 1
In file included from C:\users\ffnull\.platformio\packages\framework-arduinostm32generic\STM32\system\STM32F4\HAL_Inc/stm32f4xx_hal_conf.h:254:0,
from C:\users\ffnull\.platformio\packages\framework-arduinostm32generic\STM32\system\STM32F4\HAL_Inc/stm32f4xx_hal.h:46,
from C:\users\ffnull\.platformio\packages\framework-arduinostm32generic\STM32\cores\arduino\stm32/stm32_def.h:20,
from C:\users\ffnull\.platformio\packages\framework-arduinostm32generic\STM32\cores\arduino/Arduino.h:179,
from Marlin\src\HAL\HAL_STM32F4\HAL.h:42,
from Marlin\src\HAL\HAL_STM32F4\HAL_timers_STM32F4.cpp:29:
Marlin\src\HAL\HAL_STM32F4\HAL_timers_STM32F4.cpp: In function 'void HAL_timer_start(uint8_t, uint32_t)':

C:\users\ffnull\.platformio\packages\framework-arduinostm32generic\STM32\system\STM32F4\HAL_Inc/stm32f4xx_hal_rcc.h:488:41: error: 'SET_BIT' was not declared in this scope
SET_BIT(RCC->APB1ENR, RCC_APB1ENR_TIM5EN);\
^
Marlin\src\HAL\HAL_STM32F4\HAL_timers_STM32F4.cpp:78:7: note: in expansion of macro '__HAL_RCC_TIM5_CLK_ENABLE'
__HAL_RCC_TIM5_CLK_ENABLE();
^~~~~~~~~~~~~~~~~~~~~~~~~
Marlin\src\HAL\HAL_STM32F4\HAL_timers_STM32F4.cpp:89:7: error: '__HAL_RCC_TIM7_CLK_ENABLE' was not declared in this scope
__HAL_RCC_TIM7_CLK_ENABLE();
^~~~~~~~~~~~~~~~~~~~~~~~~
Marlin\src\HAL\HAL_STM32F4\HAL_timers_STM32F4.cpp:89:7: note: suggested alternative: '__HAL_RCC_TIM4_CLK_ENABLE'
__HAL_RCC_TIM7_CLK_ENABLE();
^~~~~~~~~~~~~~~~~~~~~~~~~
__HAL_RCC_TIM4_CLK_ENABLE
Marlin\src\HAL\HAL_STM32F4\HAL_timers_STM32F4.cpp:90:53: error: 'TIM7' was not declared in this scope
timerConfig[1].timerdef.Instance            = TIM7;
^~~~
Marlin\src\HAL\HAL_STM32F4\HAL_timers_STM32F4.cpp:90:53: note: suggested alternative: 'TIM1'
timerConfig[1].timerdef.Instance            = TIM7;
^~~~
TIM1
Marlin\src\HAL\HAL_STM32F4\HAL_timers_STM32F4.cpp:94:31: error: 'TIM7_IRQn' was not declared in this scope
timerConfig[1].IRQ_Id = TIM7_IRQn;
^~~~~~~~~
Marlin\src\HAL\HAL_STM32F4\HAL_timers_STM32F4.cpp:94:31: note: suggested alternative: 'TIM5_IRQn'
timerConfig[1].IRQ_Id = TIM7_IRQn;
^~~~~~~~~
TIM5_IRQn
*** [.pioenvs\steval3dp\src\src\HAL\HAL_STM32F4\HAL_timers_STM32F4.o] Error 1
```

thinkyhead, 2018-06-08:
Looks like a question for @Morten

ffnull, 2018-06-27:
Sorry for delay. Compilling without issues. Continuing work on board. Will report of news

ffnull, 2018-06-27:
Ok. Redefined pins. Compiled ok. But im stuck again. Im have on board L6474 but in Marlin configuration is inly L6470. And i cant uderstand how to property set pins for this drivers

xC0000005, 2018-06-27:
I don’t have those drivers, but looking at the library, it looks like you first uncomment the <motor>_IS_L6470 settings.
Then in the pins, you set the enable pin as the CS pin for that particular stepper. I’d try it and see what happens. There’s an arduino library for 6474 here:
https://github.com/MotorDriver/L6474 <https://github.com/MotorDriver/L6474>  which you might be able to make work with STM32GENERIC.

xC0000005, 2018-07-10:
I believe the right way to configure it is in the pins file for the board, set the enable pin for each driver to the CS pin for that driver.

ffnull, 2018-07-14:
Nope.Stuck on adding library. Library added but platformio doesnt see it. Copy files manual, but its ignore header file
`Compiling .pioenvs\steval3dp\src\src\HAL\HAL_STM32Fx\HAL_spi.o
Compiling .pioenvs\steval3dp\src\src\HAL\HAL_STM32Fx\HAL_timers.o
Compiling .pioenvs\steval3dp\src\src\HAL\HAL_STM32Fx\L6470.o
Marlin\src\HAL\HAL_STM32Fx\L6470.cpp:12:10: fatal error: L6470.h: No such file or directory

***************************************************************
* Looking for L6470.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:L6470.h"
* Web  > https://platformio.org/lib/search?query=header:L6470.h
*
***************************************************************

#include <L6470.h>
^~~~~~~~~
compilation terminated.
*** [.pioenvs\steval3dp\src\src\HAL\HAL_STM32Fx\L6470.o] Error 1`

xC0000005, 2018-07-15:
If you copy the library into either STM32GENERIC’s libraries folder or somewhere under marlin, does it compile? If not, it’s probably the library’s properties. Ones that use a properties file are (mistakenly) treated as “arduino” only.

ffnull, 2018-07-15:
Im already try it. But still sme result

ffnull, 2018-07-15:
Platformio ignore l6470.h file

ffnull, 2018-07-15:
It all does not make sense. im compile without this driver and upload to the board. CPU doesnt start at all. Because we start again from the beginning

xC0000005, 2018-07-15:
The eval has a built in st link. What is happening when it starts up?  Adding new hardware support often requires debugging. The good news is that st link debugging is usually solid.

ffnull, 2018-07-15:
Im total noob. What i need to do to provide you more info?

xC0000005, 2018-07-15:
It’s not an issue of providing me more info - debugging startup means attaching platformio or gdb/ocd to the chip and seeing what goes wrong. It could be a number of things, but given that the firmware loads at 0x8000000, at least you aren’t dealing with a bootloader. If you are using an st-link to upload your firmware you’re not far off. If you edit your platformio.ini and set the eval board as the default, you could try clicking the debug button. If all works it will attach to your board and you can look at the assembly that runs during startup, followed by C global variable init, followed by variant init, and so on. 

If you want to add support for new hardware, some level of hardware interaction is usually required (unless the new hardware is really old hardware with a different name).

ffnull, 2018-07-15:
Can you give me your email for more operative conversation and not spaming here? mine is djmorganfreeman@gmail.com

xC0000005, 2018-07-16:
> Can you give me your email for more operative conversation and not spaming here? mine is djmorganfreeman@gmail.com <mailto:djmorganfreeman@gmail.com>

Done. We can discuss how to move forward there, but there’s going to be some debugging.

ffnull, 2018-07-16:
Maybe its can help. Here is original firmware for this board, based on old marlin https://github.com/St3dPrinter/Marlin4ST

xC0000005, 2018-07-16:
Yes, this is helpful - I have a copy. What we want to do now is know what happens on YOUR board with YOUR firmware. I mailed you through email as well, but if you want to continue this way, does PlatformIO successfully flash and attach to the board when you click debug?

ffnull, 2018-08-02:
Hi there! Is any news? Did my link helps?

xC0000005, 2018-08-22:
> abadoned?

I don’t have a STEVAL-3DP board, so any more progress will have to be made with you debugging it (or someone with one debugging it). I’m sorry - there’s a limit to how much I can do via email. Perhaps someone else will know more and be able to help.

RedlineGT, 2019-02-12:
@ffnull any luck or progress? Just curious

boelle, 2019-02-20:
@ffnull status?

ffnull, 2019-02-21:
@boelle no status at all. board compiles but there problem with motor drivers

boelle, 2019-02-21:
so not a problem with marlin ?

closed 2019-03-03 by thinkyhead