MarlinFirmware/Marlin #10740: Issue with print slow down/freezing

AxMod3DPrint, 2018-05-14:
### Description

When printing the printer will slow down or come to a almost standstill and show processing.. in the console, this generally happens after the extruder is Zeroed out with G92E0 in the GCode.  Marlin 1.1.8. Freshly downloaded and configured firmware. (Configs linked via pastebin below). It also seems to not happen after a ZHop. The severity varies from a small judder after a travel move to a complete stop. The firmware seems to be calculating something, I'm just not sure what and it's taking up all the processor time.

Printer is Running a RAMPS 1.4/Arduino stack.

### Diagnostic steps taken

* Changed out Arduino to known good - no change
* Upped buffer values in Configuration_adv to check if buffer on Arduino is having issues  (changes reverted) - no change 
* Printed from both USB and SD Card to check if it is only happening on USB - no change
* Reflashed firmware from multiple machines using different USB Cables - no change
* Downloaded fresh firmware and reconfigured - no change

### Steps to Reproduce

Not quite sure how to reproduce, as it's been like this since first re-configuration. 

**Expected behavior:** Smooth printing with no stops or slow downs

**Actual behavior:** Printer slows down/judders/stops completely after a G92E0, and shows processing in the console. This does not happen when a Z Hop is involved. 

#### Additional Information

[Configuration.h](https://pastebin.com/evSL7bR1)
[Configuration_adv.h](https://pastebin.com/9wCcphEW)

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

AxMod3DPrint, 2018-05-14:
Further info:
* Example of console output - Printer slows down to an almost stop until next move
`Send: N3171 G92 E0*115`
`Recv:  T:229.94 /225.00 B:94.87 /95.00 @:0 B@:85`
`Recv: echo:busy: processing`
`Recv:  T:229.75 /225.00 B:94.90 /95.00 @:1 B@:71`
`Recv: echo:busy: processing`
`Recv:  T:229.31 /225.00 B:95.20 /95.00 @:3 B@:7`
`Recv: echo:busy: processing`
`Recv:  T:228.25 /225.00 B:95.11 /95.00 @:17 B@:36`
`Recv: echo:busy: processing`
`Recv:  T:227.50 /225.00 B:94.74 /95.00 @:21 B@:111`
`Recv: echo:busy: processing`
`Recv:  T:226.31 /225.00 B:95.05 /95.00 @:34 B@:35`
`Recv: X:99.27 Y:107.66 Z:0.45 E:0.00 Count A:68138 B:-764 Z:179`

It seems to take a while to get the `Count A:68138 B:-764 Z:179` part out on the console

thinkyhead, 2018-05-14:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

AxMod3DPrint, 2018-05-14:
@thinkyhead - It seems to have improved but at the start of the print it is still slow to get to the initial start point from home and there are still slow downs

thinkyhead, 2018-05-16:
Try some controlled tests with simple objects and take notes on when and where these slowdowns happen. And then post the G-code that produces the weird behavior so we can try it out too.

AxMod3DPrint, 2018-05-16:
I have been doing a few tests and it seems it's a Slicer issue, S3D produces these slow downs, where Cura doesn't. As such, closing the issue as it's not a problem with Marlin.

closed 2018-05-16 by AxMod3DPrint

thinkyhead, 2018-05-16:
Could I see the weird G-code and the good G-code, just for comparison? I'd love to know what S3D is doing differently.

AxMod3DPrint, 2018-05-16:
Attached in Zip format.. 

[GCode.zip](https://github.com/MarlinFirmware/Marlin/files/2008426/GCode.zip)

thinkyhead, 2018-05-24:
Try S3D again sometime, but disable this option:
[image]

thinkyhead, 2018-05-25:
AFAIK a sketchy LCD shouldn't have any effect. Marlin just sends bytes over SPI at a fixed rate and doesn't really know or care if the display receives them.

AxMod3DPrint, 2018-05-31:
Found time to set the extruder to relative, no change.

'Eagles may soar, but Weasels don't get sucked into Jet engines'

thinkyhead, 2018-06-01:
We're continuing to refine the planner. Give the bugfix branches another test in a day or two and see if they've improved any, even with S3D.

And if you do find any options in S3D that provide a workaround, we'll definitely want to know more. Sometimes it's the slicer trying to be "too clever" by overriding the jerk and acceleration in the firmware, or otherwise conflicting with the firmware configuration, but sometimes it points to an edge case that we can fix.

AxMod3DPrint, 2018-06-04:
@thinkyhead will do. It does seem like it's something that S3D is doing
which is causing the CoreXY kinematics to do strange things. The other
straight Prusa Style Cartesian is fine.