MarlinFirmware/Marlin #10741: [Q] Send ok after the moving is done.

coolzoom, 2018-05-14:
Hi Everyone, for current firmware, if i send a gcode for example G0 X100 Y100 Z10, it will response "ok" right after it receive the message.
However, i wish to receive ok once the moving is done(same as G28 the homing process).
the reason i do this is I'm trying to tell the system to do something else once it reach to the destination.

i see the G28 command is using similar method, the controller send "busy: processing" while it is still in the move. i wish i can do the same for G0 and G1 command. i did several try and only found "KEEPALIVE_STATE(IN_PROCSSS);" may related to that, not sure how i can do what i want.
can you please help me to check how to modify the code to match what i need?

GMagician, 2018-05-14:
I think this will be hard to achieve. As I know movement received are inserted in a queue and then, after some time, executed. Different is G28 that is executed sequentially and immediately.
G28 can send message as soon as it terminate while movements are terminated inside an interrupt (I suspect) and such isr requires to be executed as fastest as possible and communication can't be done there.

thinkyhead, 2018-05-16:
Try this:
```
G1 X100 F1000
M400
M117 moved!
```

thinkyhead, 2018-05-22:
I knew `M400` had to be good for something. Cheers.