MarlinFirmware/Marlin #10743: Strange values / lost a lot of GCODE commands

ikarisan, 2018-05-14:
Hi!

To get rid of my other problems I flashed the latest bugfix-1.1.x version ( 29080cefd97171cc632262aa446db6cf6c80ae7a ) to my MKS1.4 board. But now a lot of command are not working any longer and the values shown on M503 are very strange!

Send: M851
Recv: echo:Unknown command: "M851"
Recv: ok

Send: M503
Recv: echo:Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z4000.00 E500.00
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X300.00 Y300.00 Z5.00 E25.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X3000 Y3000 Z100 E10000

WTF?! M92 values with Z=4000 and E=500?!
Acceleration on X and Y at 3000?!

thinkyhead, 2018-05-14:
I guess you noticed this longstanding config default, inherited from Ultimaker.
```cpp
#define DEFAULT_ACCELERATION          3000    // X, Y, Z and E acceleration for printing moves
#define DEFAULT_RETRACT_ACCELERATION  3000    // E acceleration for retracts
#define DEFAULT_TRAVEL_ACCELERATION   3000    // X, Y, Z acceleration for travel (non printing) moves
```

ikarisan, 2018-05-15:
Sorry, I had to close this. 
For unknown reasons GIT had overwritten my configuration.h file. Normally this should not happen if the local file is modified. After restoring my last configuration backup and reflashing, everything works fine again.