MarlinFirmware/Marlin #10749: [bugfix-1.1.x] Dual X carriage kinematics completely broken

orcinus, 2018-05-15:
Setup:
- DUAL_X_CARRIAGE
- UBL manually leveled and active
- bed size 200x300, 0 at (0,0) corner of the bed
- X2_MIN_POS 80 (initial "safe" setting)
- X2_MAX_POS 295
- DXC_AUTO_PARK_MODE
- TOOLCHANGE_PARK_ZLIFT   0.2
- TOOLCHANGE_UNPARK_ZLIFT 1

I'm noticing multiple issues. It looks like the planner is completely confused by the tool changes. I am not sure if this is due to UBL or a general issue with dual x support. Will test without UBL tomorrow.

Issues:
1. with hotend Z offset set up, every tool change to T1 appears to "shave off" the Z offset, but does not "restore" it on tool change back to T0 - in time, this results in a hotend crashing into the bed
2. after every tool change, the first G1 that follows (whatever it is, even if not affecting X) results in the carriage going to its X parking position *very* slowly, then the actual requested G1 move after that
3. i've measured the X hotend offset of +30.something with the X2_MIN and X2_MAX set up as mentioned above - when setting the offset to that value, the next T1 attempts to exit parking position (295) *in the wrong direction* (i.e. +X) and slams the carriage into the endstop

I haven't got the first clue what's going on here...
Any help would be appreciated.

orcinus, 2018-05-15:
Issue #1 happens without Z offset too (just checked).

orcinus, 2018-05-15:
Issue 1 *is* UBL specific (just confirmed by disabling UBL).
Issue 2 *is not* UBL specific.

Additional DXC_AUTO_PARK_MODE issue. In this mode, T0 and T1 will unpark the *next* tool to *current* tool's X, Y and Z. This behaviour doesn't make a lot of sense, and isn't very safe either, considering the tools ranges of motion do not completely overlap. The tools should either unpark to their respective last coordinates, or (easier) not unpark until the next G1.

orcinus, 2018-05-15:
Issue 3 *is not* UBL specific.

Also, setting the X offset to a negative value does absolutely nothing. 
It behaves exactly the same as setting it to 0. I'm completely baffled.

It's 5:40am, so i think i'm going to continue poking around tomorrow, i can't think straight anymore.

orcinus, 2018-05-15:
Issue 3 revelation:

A-HA! I somehow missed the comment that says that hotend X offset adjusts the X2_HOME_POS in DUAL_X_CARRIAGE mode, not the actual offset. Well that explains it. Even if it's not a very logical behaviour.

Also, why are X2_MAX_POS and X2_HOME_POS separate parameters? Is there an example of a mechanical setup where the Xmax endstop is *not* at the Xmax of the axis?

thinkyhead, 2018-05-15:
> Issue 1 is UBL specific (just confirmed by disabling UBL).

Yeah, we need to improve the integration of UBL to be able to deal with Z hops. The same issue appears to exist with `G10`/`G11` Z changes, we think due to UBL having isolated movement planning that isn't integrated into the `Planner` class like MBL and ABL. The `SEGMENT_LEVELED_MOVES` option might fix the issue because it doesn't use UBL's isolated movement method. If that fixes the issue, then it just reminds us that we have work to do to get UBL better integrated.

> hotend X offset [sets] the `X2_HOME_POS` in `DUAL_X_CARRIAGE`

Yes, that's important to know! It's just the way it's always been. I'm glad you sorted out Issue 2.

> why are `X2_MAX_POS` and `X2_HOME_POS` separate parameters?

Likely because in-general the min/max positions and home positions are separate parameters for all axes. The max pos defines the farthest point the carriage can physically reach, while the home pos defines the position to set when homing to an endstop. At the endstop end of an axis, they are always the same value.

AFAIK for Dual X Carriage these values are always going to be the same. But semantically, throughout Marlin code generally, they still maintain separate meanings. When we're doing code related to homing we use one, and when dealing with limits we use the other.

---
Regarding Issues 1 and 2, it would be great to have your help troubleshooting. None of the core developers has a Dual X Carriage setup, so regressions and undiscovered issues like this are all too common.

orcinus, 2018-05-15:
> Regarding Issues 1 and 2, it would be great to have your help troubleshooting. None of the core developers has a Dual X Carriage setup, so regressions and undiscovered issues like these are all too common.

Poked through `tool_change` and `dualx_tool_change` a bit yesterday, but couldn't find anything obviously out of place before i fell asleep. Will enable debug info later today and see if anything useful comes up.

orcinus, 2018-05-16:
Did some more poking around, found nothing.
It happens in ABL, MBL and UBL, so seems unrelated to bed leveling.

The symptoms look like there's a stray planner.buffer_line that doesn't get executed straight away, or a misplaced set_destination_from_current or something, but i couldn't find anything obviously out of place in Marlin_main.

The next step would be to revert tool_change to an older release's, see if it's still happening (to pinpoint where the issue's coming from), then work forwards through releases to figure out where the regression came from.

Unless someone has a better idea : /

orcinus, 2018-05-17:
Well, i've got bad news, and good news.

The bad news is: i still haven't found the exact code causing the regression.
The good news is: i still haven't found the *exact* code causing the regression.

But it is somewhere in tool_change. I've transplanted the tool_change from 1.1.0RC8 (pre-split into mixing_tool_change, dualx_tool_change and parking_extruder_tool_change) into 1.1.x-bugfix, and the issue is gone.

Now i need to trawl through the diff between them and figure out what exactly is functionally different...

bruce356, 2018-05-17:
Hi not sure what the problem is but I use bugfix v1.1x  dual x carriage with RUMBA board and it works perfectly for me.
Obviously my printer size would be different to yours and I have changed a couple of Pins settings from standard RUMBA (broken connector) but I have attached my config files so you can make some comparisons with your configs
[Marlin Sony Create-X2 2nd TMC2100_X-axis & Extruders_only v1.1.x_03-04-2018_USED.zip](https://github.com/MarlinFirmware/Marlin/files/2012393/Marlin.Sony.Create-X2.2nd.TMC2100_X-axis.Extruders_only.v1.1.x_03-04-2018_USED.zip)
 and maybe find out why its not working for you.

orcinus, 2018-05-17:
@bruce356 you seem to have checked out the branch *right* before the `tool_change` methods got changed in https://github.com/MarlinFirmware/Marlin/pull/10243/

So that likely explains why it's working fine for you.
I'll check out from a commit right before that PR got merged to verify just in case, though.

orcinus, 2018-05-18:
I've found the culprit for issue 2!

This:
```c++
          // Move back to the original (or tweaked) position
          do_blocking_move_to(destination[X_AXIS], destination[Y_AXIS], destination[Z_AXIS]);
```

causes the issue described (super-slow move to home on first G1 after a tool change). Replacing it with:

```c++
          // Move back to the original (or tweaked) position
          prepare_move_to_destination();
```

Out of curiosity, i've tried the config from @bruce356 since we have a very similar configuration (almost the same bed size too), and the issue is present with his configuration as well with `do_blocking_move_to`.

Since i haven't been following Marlin internals for years, i've got no idea what the difference between `do_blocking_move_to(<destination>)` and `prepare_move_to_destination()` is and whether this is intentional or not. What i can say is that printing with every tool change causing an unnecessary super-slow crawl home isn't very efficient :)

@thinkyhead i know you've said (somewhat rudely) you're busy and want to be left alone in another issue, but can you chime in re: `do_blocking_move_to(<destination>)` vs. `prepare_move_to_destination()`?

orcinus, 2018-05-18:
I've identified another problem with DUAL_X_CARRIAGE - odd homing behaviour with T1 selected - but i want to confirm it isn't due to a bad config first. I'll create a separate issue for it when confirmed.

orcinus, 2018-05-19:
Can't replicate the homing issue, will get back to that if it ever happens again.
Gonna test issue 1 with SEGMENT_LEVELED_MOVES.

orcinus, 2018-05-19:
SEGMENT_LEVELED_MOVES *does not* help.

thinkyhead, 2018-08-13:
Moving discussion to #11412

closed 2018-08-13 by thinkyhead