MarlinFirmware/Marlin #10753: define USE_ZMAX_PLUG stops extruding filament

dmitrygorbenko, 2018-05-15:
### Description

The USE_ZMAX_PLUG directive stops extrude motor.

### Steps to Reproduce

1. have a Z_MAX endstop on a printer
2. set "#define EXTRUDE_MINTEMP 5" (to let the extrude motor rotate without heating a nozzle)
3. set "#define USE_ZMAX_PLUG"
4. compile and upload the firmware (marlin 1.1.8 or the master branch)
5. make the Z_MAX endstop being triggered (bed reaches Z_MAX position)
6. insert a filament into extruder and notice filament's position
7. connect to the marlin by usb cable (I used Pronterface program)
8. run the command `G1 F100 E100`
9. run the command `M114` -- make sure marlin says E100
10. run the command `G1 F100 E0`
11. run the command `M114` -- make sure marlin says E0

**Expected behavior:** The filament returns into the same position as it was in the beginning.

**Actual behavior:** The filament goes back a half way (50mm instead of 100mm), so you need to run the command `G1 F100 E-100` to get to its initial position.

#### Additional Information

* Marlin thinks it pulled filament backward like I told him (actually it did only half way) -- see console dump for details.
* If you run the same test on speed F200 the bug disappears.
* If you raise your bed to un-trigger Z_MAX endstop the bug disappears.
* The bug happens also if I run "E-100" and then "E0" 0 filament stays at "-50" while marlin thinks it pushed it to 0.
* Sometimes pulling backward happens with the same length I sent by the command (it's like 5% of tests I performed)

My config: [config.tar.gz](https://github.com/MarlinFirmware/Marlin/files/2005766/config.tar.gz)
Demonstration of the issue: https://www.youtube.com/watch?v=79K0k57sRrI

The Pronterface's console dump:
```
Connecting...
start
Printer is now online.
echo:Marlin 1.1.8
echo: Last Updated: 2017-12-25 12:00 | Author: Dima Gorbenko (dmitriy.gorbenko@gmail.com)
echo:Compiled: May 12 2018
echo: Free Memory: 3614  PlannerBufferBytes: 1264
echo:Hardcoded Default Settings Loaded
echo:  G21    ; Units in mm
echo:  M149 C ; Units in Celsius
echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z320.00 E190.40
echo:Maximum feedrates (units/s):
echo:  M203 X400.00 Y400.00 Z100.00 E100.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X1000 Y1000 Z200 E8000///
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P1500.00 R2000.00 T2000.00
echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z1.00 E5.00
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Material heatup parameters:
echo:  M145 S0 H180 B70 F0
echo:  M145 S1 H245 B110 F0
echo:PID settings:
echo:  M301 P22.20 I1.08 D114.00
echo:SD card ok
>>> G1 F100 E100
SENDING:G1 F100 E100
echo:endstops hit:  Z:0.00
>>> M114
SENDING:M114
X:0.00 Y:0.00 Z:0.00 E:100.00 Count A:0 B:0 Z:0
>>> G1 F100 E0
SENDING:G1 F100 E0
echo:endstops hit:  Z:0.00
>>> M114
SENDING:M114
X:0.00 Y:0.00 Z:0.00 E:0.00 Count A:0 B:0 Z:0
```

thinkyhead, 2018-05-16:
Sounds very odd. Here are some things you might try:

- Try disabling `USE_***_PLUG` for the endstops you don't have.
- Try setting `MINIMUM_STEPPER_PULSE` to 2.

And please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch.

dmitrygorbenko, 2018-05-16:
Ok, I will perform tests soon (today or tomorrow)
thanks

dmitrygorbenko, 2018-05-17:
Hi,

> M502, M500

I've got next while executing `M502` and `M500`:
```
>>> M502
SENDING:M502
echo:Hardcoded Default Settings Loaded
>>> M500
SENDING:M500
Error:EEPROM disabled
[ERROR] Error:EEPROM disabled
```

> Try disabling USE_***_PLUG for the endstops you don't have.

I have all 6 endstops, so there is nothing to disable

> Try setting MINIMUM_STEPPER_PULSE to 2.

I've changed this variable from 0 to 2, uploaded firmware, conduct the test, and got the same error:
```
>>> G1 F100 E100
SENDING:G1 F100 E100
echo:endstops hit:  Z:0.00
>>> M114
SENDING:M114
X:0.00 Y:0.00 Z:0.00 E:100.00 Count A:0 B:0 Z:0
>>> G1 F100 E0
SENDING:G1 F100 E0
echo:endstops hit:  Z:0.00
>>> M114
SENDING:M114
X:0.00 Y:0.00 Z:0.00 E:0.00 Count A:0 B:0 Z:0
```
Marlin thinks it pushed filament back to 0, but he did half way of 100.

Any other tests to do ?
I'll try to make them more quickly.

Thanks,
Dima

dmitrygorbenko, 2018-05-17:
hold on a second... I forget to test against different branches... stay tuned

dmitrygorbenko, 2018-05-17:
branch `bugfix-1.1.x` works as expected. No messages on console about Z endstop.
branch `bugfix-2.0.x` works as expected. No messages on console about Z endstop.

btw, I LOVE structure of 2.0.x branch, it's so easy to edit files and find them.
and adding CMake is nice plus as well, thank you !

do you recommend to use `bugfix-1.1.x` ?

thinkyhead, 2018-05-25:
I like the `bugfix` branches and I use them (naturally). We hear there are issues with lost steps affecting some stepper drivers with some setups (and hope to track that down soon!) but I personally have not seen the issue on any of my machines. If `bugfix-1.1.x` works for you, then continue using it. It's the best Marlin so far.

dmitrygorbenko, 2018-05-30:
Thank you.
I think we can close the issue, right ?

closed 2018-05-30 by dmitrygorbenko