MarlinFirmware/Marlin #10755: [2.0.x, HAL] Question: What is "HAL_DUE/usb/sd_mmc_spi_mem.cpp" for?

marcio-ao, 2018-05-15:
As I was working on getting USB thumb drive support implemented, I ran into a peculiar situation where my USB thumb drive code was working fine on an Atmega2560 EinsyRambo, but as soon as I compiled it for Archim using the atsam3x8e, the board would appear to freeze up when I inserted a thumb drive.

After some investigation, I found out that this was due to continuous accesses to the flash drive which were not happening on the Atmega2560 chip. It turns out the function "[sd_mmc_spi_usb_read_10](https://github.com/MarlinFirmware/Marlin/blob/0ea60d506a9d5676afe0a47c39dea32411068f1b/Marlin/src/HAL/HAL_DUE/usb/sd_mmc_spi_mem.cpp#L61)" was getting called repeatedly even when Marlin was not actively reading from the card.

When I modify that function to return CTRL_NO_PRESENT immediately, everything starts to work fine and my USB code behaves as it should.

So, my question is, what is "sd_mmc_spi_usb_read_10" and what is it supposed to be doing? Also, what calls it? I can see that it is getting called repeatedly, but yet there are no calls in Marlin to it. Why does this only exist on the atsam3x8e HAL and not the Atmega2560 HAL?

Any answers would be very much appreciated. Thank you!

-- Marcio

p3p, 2018-05-15:
I've would assume it is part of the DUEs USB MSD implementation, so when the native usb port in used with a pc the sd card shows up as a usb disk drive.

p3p, 2018-05-15:
Afraid I haven't got a Due based board set up so I can't comment on its performance, the LPC176x implementation works fine for me but I implemented that one and it uses a dedicated onboard sdcard that Marlin cannot access directly atm, needing to use spi in an interrupt makes it complicated to inter operate with Marlin code.

ejtagle, 2018-05-16:
@marcio-ao : When plugged in the printer into an USB port, Windows (/Mac/Linux) detects the MSD and reads the whole FAT. That is why i had to speed up the SD card access as much as possible. The SPI routines are able to transfer at 25Mhz (25 megabits) with no problems, even the Software based ones.
The printer freezes for maybe 1/2 seconds. But this only happens when the printer is idle, and not printing. If the printer is printing, then the PC is not allowed to access the SD at all.

ejtagle, 2018-05-16:
At some point i did a throttling of the reads, but that reduces the speed of the transfers from the PC to the SD card... 
Right now, while copying files from the PC to the SD, i get 700Kbytes/second of transfer speed.

closed 2018-07-11 by marcio-ao