MarlinFirmware/Marlin #10757: Problem to heat hot-bed purse i3 marlin

makerPeppe, 2018-05-15:
I have to set up the pid for hot bed of my prusa i3.
I send at my prusa i3 by repetier-host the m303 e-1 c8 s100 for pid autotune.
However my 3d printer stopped the heating after about 10 minute because of the error(the temperature arrived at 94°C ):
"PID Autotune failed! timeout"

At the bed there was 10,9 V instead of 12V.
This bed must have 12 V. Can it be the problem?
I inserted also under the bed a sheet of aluminum foil to retain heat but I didn't resolve the problem. How can I do?

Grogyan, 2018-05-15:
The power supply has a little potentiometer which sets the output voltage of the power supply.
This can sometimes help to turn up the voltage by turning the pot.

makerPeppe, 2018-05-16:
I used an ATX for 500W. Can it be possible?

boelle, 2018-05-16:
have you put insulation under the bed? maybe heatloss is to great to get to 100

10.9V is not the problem

makerPeppe, 2018-05-16:
Yes I have put insulation.
What are the other material to insulate thermally the bed?
What are the other solution?

thinkyhead, 2018-05-24:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. For best results getting help with configuration and troubleshooting, please use the following resources:
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.

closed 2018-05-24 by thinkyhead