MarlinFirmware/Marlin #10761: 1.1.8-bugfix: "Fade height" on LCD wrong

insanity67, 2018-05-16:
Hi!
I have a problem, and dont know if this is a bug or ok: Fade height shown by M503 is 2mm, on LCD showing 0mm. If I change this on the LCD to 1mm, M503 shows 1mm.
When I save this, and restart printer, M503 shows 1mm, but again 0mm on the LCD.

Best regards!

thinkyhead, 2018-05-16:
Sounds like a potential bug. We'll look into it!

thinkyhead, 2018-05-16:
>  on LCD showing 0mm

I assume you mean, when you go to the "Tune" menu or the "Bed Leveling" menu and want to edit the value, it shows 0 as the current edit value, yes?

insanity67, 2018-05-16:
Yes, the current value shown is always 0.00 after restart/reset of the printer. Looks like the value is read from EEPROM at boot (because showing in M503) but somehow the LCD uses another variable to show that is not populated?!

thinkyhead, 2018-05-16:
And you are using the latest `bugfix-1.1.x` branch? So far as I can tell, in the current code that temporary variable _is being set_ prior to showing the edit item. However, if it's set with `M420 Znnn` then you have to exit and re-enter the menu for it to update on-screen.

insanity67, 2018-05-16:
Yes the latest branch, pulled yesterday.
OK thanks, I have to test if that happened with M420 Z.

But, would it not be better to also show the value entered with M420 Z on LCD?

insanity67, 2018-05-17:
Finally had some time to test:
When I power on my printer, in the menu under "tune" the "Fade Height" is 0.00.
As soon as I start a print (even while preheating, no `M420` in the GCode) the "Fade Height" is showing the correct value from EEPROM.

Dont know if this is a bug, or a wanted behaviour? I would find it better if the value was there after I start the printer.

thinkyhead, 2018-05-20:
Ok, so it's ok in the Tune menu but not ok in the Prepare menu.

insanity67, 2018-05-20:
Yes, Prepare Menu shows 0.0 right after printer power up. As soon as I start the print Prepare changes to Tune, and there is my Value from EEPROM.

thinkyhead, 2018-05-20:
Ok, I've patched the two bugfix branches to fix the issue. Should work better now.

insanity67, 2018-05-20:
Compiled, tested -> works! Thanks for your great support!

commit 68d7fcec (2018-05-20): Simplify and fix Z fade height edit

Fix #10761
commit bf61be8e (2018-05-20): Simplify and fix z fade height editing

Fix #10761
closed 2018-05-20 by insanity67