MarlinFirmware/Marlin #10762: Swap TEMP_BED_PIN

crashoverdone, 2018-05-16:


GMagician, 2018-05-16:
Theoretically it should be possible. What If you connect to T2 your working sensor (the one in T0) maybe you have some other issue on T1 input

crashoverdone, 2018-05-16:
GMagician thank you for your suggestion but i've tried that too.
both sensores are ok only the T1 not ok and changing the pins_RAMPS.h as i mentioned before gives a bad reading on T2. I dont know if its needed to change other file or just this one.

crashoverdone, 2018-05-17:
GMagician maybe you're right. Is there a way to measure to check that?

GMagician, 2018-05-17:
You can try to disconnect all sensor and match read values.
I'm almost sure that changing pins in .h is enough. You can try also moving T2 input to T0 and see if it works. This way you confirm that pin swapping is working.

crashoverdone, 2018-05-19:
GMagician, hi and thanks a lot for your tips.
My ramps its f...ed so i order a  new one and all fine.

GMagician, 2018-05-19:
Happy to have helped you and a little sorry for your board...

Now you can close this, I think

closed 2018-05-20 by thinkyhead

JoHo82, 2019-11-26:
Hi.
I stuck in a problem with my Ender 2
.
I demaged the Pin A6 (long story short: I pressed the screws on the Heatbed and so made a short circuit between 12V and the TempBedPin...)

So I now try to move the TempBedPin to A5 (which is free at my board). I soldered it to the pin.
Then I changed #define TEMP_BED_PIN in "pins_SANGUINOLOLU_11.h" from 6 to 5:

#define TEMP_0_PIN 7 // Analog Input (pin 33 extruder)
#define TEMP_BED_PIN 6 // Analog Input (pin 34 bed)

to

#define TEMP_0_PIN 7 // Analog Input (pin 33 extruder)
#define TEMP_BED_PIN 5 // Pin swapped from 6 to 5

But it doesn't work. It shows me constantly 44°C (what is different to 110°C that it showed when it was set to A6)

What is wrong? do i need to make additional settings?

Thanks!

shitcreek, 2019-11-26:
My guess is that pin 6 has a circuit whereas pin 5 is just a regular analog pin. https://reprap.org/mediawiki/images/8/83/Sanguinololu-schematic.jpg