MarlinFirmware/Marlin #10763: Unable to compile with NO_LCD_MENUS enabled

Looxoor, 2018-05-16:
### Description

Downloaded the latest vesion of bugfix 1.1.x, tried to compile with NO_LCD_MENUS (issue #10578), it failed.

### Steps to Reproduce

Tried to compile it (downloaded the latest bugfix an hour ago) with the "NO_LCD_MENUS" option for the first time today and got two errors during compilation:

First, in "G26_Mesh_Validation_Tool.cpp":
```
'is_lcd_clicked' was not declared in this scope
```

I commented out the G26 option in configuration.h and tried to compile again,
then the next compile time error occured:

This time in temperature.cpp
```
'ADC_KEYPAD_PIN' was not declared in this scope
```

So I had to comment out no lcd menus for now and use slim lcd menus instead.

Hope this helps.

**Expected behavior:** [What you expect to happen]
That the compilation succeeds.

**Actual behavior:** [What actually happens]

It fails.
#### Additional Information
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2010668/Marlin.zip)

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

thinkyhead, 2018-05-18:
Thanks for the report. I'll patch it up asap.

thinkyhead, 2018-05-19:
These should be cleared up now.

closed 2018-05-19 by thinkyhead