MarlinFirmware/Marlin #10765: First compile for MKS SBASE fails

cheffffe, 2018-05-17:
Hello, I tried my first creation of Marlin Firmware for MKS SBASE with Atom and PlatformIO. I loaded Marlin into Atom and edited configuration.h and configuration.adv.h. So far so good, but when I want to PIO Build I get always this error:

[05/17/18 08:54:25] Processing LPC1768 (platform: nxplpc; board: lpc1768)
 
Warning! `monitor_baud` option is deprecated and will be removed in the next release! Please use `monitor_speed` instead.
Warning! Library `{'requirements': None, 'name': 'CMSIS-LPC1768'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'requirements': None, 'name': 'CMSIS-LPC1768'}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
Verbose mode can be enabled via `-v, --verbose` option

PLATFORM: NXP LPC > NXP mbed LPC1768
SYSTEM: LPC1768 100MHz 64KB RAM (512KB Flash)
DEBUG: CURRENT(cmsis-dap) ON-BOARD(cmsis-dap) EXTERNAL(blackmagic, jlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(off) COMPATIBILITY(light)
Collected 9 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <CMSIS-LPC1768> v0.0.0
|-- <U8glib-HAL> v0.4
|-- <TMC2130Stepper> v2.3.0
CalledProcessError: Command 'fsutil fsinfo volumeinfo C:' returned non-zero exit status 1:
File "C:\Users\chefffe\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 172:
env.SConscript(item, exports="env")
File "C:\Users\chefffe\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "C:\Users\chefffe\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "C:\Users\chefffe\Desktop\Marlin-bugfix-2.0.x\Marlin\src\HAL\HAL_LPC1768\upload_extra_script.py", line 51:
volume_info = subprocess.check_output('fsutil fsinfo volumeinfo ' + final_drive_name)
File "C:\Users\chefffe\.platformio\python27\Lib\subprocess.py", line 219:
raise CalledProcessError(retcode, cmd, output=output)
 [ERROR] Took 0.93 seconds

cheffffe, 2018-05-17:
Thats it! Thanks a lot!

thinkyhead, 2018-05-25:
Thanks for the note! This should now be patched.

closed 2018-05-25 by thinkyhead