MarlinFirmware/Marlin #10766: Not moving servo for switching extruder

Studna3D, 2018-05-17:
The servo doesnt switch when I change extruder nozzle. The last version of the code was working.

GMagician, 2018-05-18:
Please your configuration. I have it perfectly working

Studna3D, 2018-05-18:
I use bugfix-2.0.x. and MKS Sbase. The last version Bugfix 2.0.x was working, but now the servo not moving. 
I watched the code here and saw change in tool_change.cpp 6 days ago.

GMagician, 2018-05-18:
well I'm using a "yesterday" copy of it and it's working. If you attach your configuration I can try to look which differences with mine and check involved code

Studna3D, 2018-05-18:
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2017429/configuration.zip)

GMagician, 2018-05-18:
Ah ok, I also have SWITCHING_NOZZLE not only SWITCHING_EXTRUDER. My cnf is a single stepper used for move filament and 2 nozzles switching. switching is done by a servo
What about yours configuration?

about:

> I watched the code here and saw change in tool_change.cpp 6 days ago.

if I'm not wrong developers moved some methods from one class to other

GMagician, 2018-05-18:
Are you sure you are using this cnf files?
I see `#define NUM_SERVOS 3` but just below '#define SERVO_DELAY { 300 }', it should not compile

GMagician, 2018-05-18:
by the way try to change

`#define DO_SWITCH_EXTRUDER (SWITCHING_EXTRUDER_SERVO_NR != SWITCHING_NOZZLE_SERVO_NR)
`

to

#define DO_SWITCH_EXTRUDER true

in tool_change.cpp line 350.. there are other part in code to change, I'll post a PR

Studna3D, 2018-05-19:
I change define switching extruder and switching nozzle together and it's working. 
Thank you very much!

GMagician, 2018-05-19:
@Studna3D when both SWITCH are enabled the define can be left untouched

GMagician, 2018-05-19:
I think this can be closed, also a PR has been posted and should fix it.
When merged, if you have again issues, you can reopen

closed 2018-05-25 by thinkyhead