MarlinFirmware/Marlin #10767: Bugfix 2.0 - Error handling added to build script

jokeman3000, 2018-05-17:
### Description
Recently scripts were added to automatic drive selection as part of the build process. On mac, the script fails when it encounters a drive it does not have access to.

OSError: [Errno 13] Permission denied: '/Volumes/TMBackup/':
File "/Library/Python/2.7/site-packages/platformio/builder/main.py", line 171:
env.SConscript(item, exports="env")

With the current script,the compile process stops until I unmount this volume.

Bob-the-Kuhn, 2018-05-18:
The title implies that you have a solution.  Please post it if so.

Bob-the-Kuhn, 2018-05-19:
I need your help.  I just enabled Time machine but am not getting any errors.  Right now it's doing it's first backup.  It's visible in Finder so I assume it's security isn't set properly.  What are the security settings on your drive?

FYI - I found a bug in the MAC portion of the script.  You can try the attached if you feel like it.  There is not any added error recovery in it.

[upload_extra_script.py.zip](https://github.com/MarlinFirmware/Marlin/files/2019075/upload_extra_script.py.zip)

jokeman3000, 2018-05-19:
Hey Bob, happy to help. It's a nas with time machine functionality. It is password protected but authentication is handled by time machine. 

I don't think time machine should be the focus, you would have the same issue with a samba share that requires authentication.

I was suggesting error handling because it returns that it does not have access and just quits compiling. Since the script is just scanning volumes it should move to the next one.

Bob-the-Kuhn, 2018-05-20:
I have not been able to re-produce the problem.  Most likely because I've never been in the NAS or samba worlds.

Please try the following and let me know what happens.  I've added exception handling at the directory read step.  
[upload_extra_script.py.2.zip](https://github.com/MarlinFirmware/Marlin/files/2019916/upload_extra_script.py.2.zip)

If that doesn't fix it then please post the following:
1) The contents of the build window.  There's a right click menu that will make it easy to copy everything.
2) The contents of the terminal window.  That's the window at the bottom of the Atom screen.  It'll be blank until you kill the build window.  In it you should see the following text plus, hopefully, detailed error messages.
```
build_type:   upload

Working

python version 2.7.10
build_type:   upload
starting platformio
```

jokeman3000, 2018-05-20:
Hi Bob, 

Looks like you got it. I'll test periodically to make sure but I did try it mid backup and seemed ok.

Bob-the-Kuhn, 2018-05-20:
I've moved the update into the code base via PR #10802

closed 2018-05-20 by Bob-the-Kuhn