MarlinFirmware/Marlin #10769: [2.0.X] LPC1768 - Compilation error when activating EMERGENCY_PARSER

Nutz95, 2018-05-18:
I've just pulled the latest changes from the repo (50270b53a0a2b8d716f1a771563e17f9753e14ce) (18th of may 2018).

When i enable **EMERGENCY_PARSER** from **configuration_adv.h** 
i've got some compilation error:

> Marlin\src\gcode\gcode.cpp:713:48: error: 'strchr_P' was not declared in this scope

indeed when i look inside file **Marlin\src\HAL\HAL_LPC1768\include\arduino.h**  at line 95, 
#define strlen_P strlen
is defined but not 
#define strchr_P strchr

when i look at the change history of arduino.h for LPC1768 the issue comes from commit 2242b98248a50a2a3a8a29c9ec4f40308a2a0450 from 25/04/2018  where only #define strlen_P strlen is set but not strchr_P 

Both strchr_P  and strlen_P  are used inside Marlin\src\gcode\gcode.cpp at line 713 and 714 inside function void GcodeSuite::process_subcommands_now_P(const char *pgcode)

----------------how to fix ------------------------------
In order to fix this compilation issue, 
just declare #define strchr_P strchr inside Marlin\src\HAL\HAL_LPC1768\include\arduino.h at line 96

p3p, 2018-05-18:
Thanks for submitting this, strchr_P is not needed with my config (or wasn't at the time) when the emergency parser is enabled so I didn't notice the requirement.

thinkyhead, 2018-05-20:
Patched!

closed 2018-05-20 by thinkyhead