MarlinFirmware/Marlin #10771: Dual Z Driver problem

Gigahertz77, 2018-05-18:
Marlin 1.1.16

For E1 ( Z2 ) the number sequence is this 

 *                                      X, Y, Z, E0 [, E1[, E2[, E3[, E4]]]]
 */
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 100.40, 100.45, 803.98, 200.18, 803.98 } 

or 

 *                                      X, Y, Z, E0 [, E1[, E2[, E3[, E4]]]]
 */
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 100.40, 100.45, 803.98,  803.98, 200.18,  }
??

GMagician, 2018-05-18:
There is a ',' at end DEFAULT_AXIS_STEPS_PER_UNIT tha shouldn't be there...I think this will set to 0 the value. You can repeat same value or, if missing, last E value will be used.

Don't checked code about this but read only reported defines

thinkyhead, 2018-05-20:
When using Dual Z axes they both use the value in the Z (3rd) position.

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-03-14 by thinkyhead