MarlinFirmware/Marlin #10781: [2.0.x] RE-Arm No Comms via USB, Marlin appears to boot

adamnock, 2018-05-19:
### USB Comms with Marlin 2.0.x on RE-Arm Board failing even though Marlin appears to boot.
Base Config

### Steps to Reproduce

1. RE-Arm board with RAMP's Shield removed (base hardware)
2. Compile Marlin using stock Configuration.h excepting Board define
3. Build using LPC1768 Target in PlatformIO
4. Upload to SD and reboot board (or power cycle, no difference)
5. Allow 60seconds for boot, then attempt to connect via USB Serial. Device is present (and bus reported descriptor is "Marlin USB Device")
6. No command response observed.

(extra)
Compiled again with EEPROM enabled, confirmed eeprom.dat was created, so Marlin appears to be booting.

**Expected behavior:** Should be able to issue basic commands, eg M119, and get response.

**Actual behavior:** No Comms

#### Additional Information

Ive tried this against two PC's, ruling out (i hope) software issues.
Ive also comparatively loaded Smoothieware onto the board to check for board issues. Smootie runs fine and connects fine.

This is using the as current branch of bugfix-2.0.x, no changes to the base Configuration as stated.

Squid116, 2018-05-19:
I am experiencing exactly the same symptoms described here.

I can see the firmware.bin is being renamed to firmware.cur so I assume that all checks out?

p3p, 2018-05-19:
The default configs were recently changed to disable the secondary serial port, please set Serial port 1 to the usb connection (-1) or re-enable the secondary port.

Roxy-3D, 2018-05-19:
This is where a Max7219 LED Matrix would come in handy.   They are under $2 on eBay including shipping.    (And they look cool even if you don't have a problem).

https://www.ebay.com/p/Max7219-Dot-Matrix-Module-MCU-Control-8x8-Display-Module-for-Arduino/1672606113?iid=222852955295&_trkparms=aid%3D222007%26algo%3DSIM.MBE%26ao%3D1%26asc%3D50544%26meid%3D8113e82d0f9b48208229ee5eabd17d69%26pid%3D100005%26rk%3D4%26rkt%3D12%26sd%3D222709799381%26itm%3D222852955295&_trksid=p2047675.c100005.m1851

adamnock, 2018-05-19:
@Roxy-3D What for? Showing boot status?

adamnock, 2018-05-19:
@p3p That sorted it.

Thats a rather non-intuitive change though. Given the majority of off the shelf hardware is going to have an intended port for serial comms as a primary, would it not make sense to have Serial Port 1 be defined by default at the board level, and overridable instead of a main config item?

thinkyhead, 2018-05-25:
> would it not make sense to have Serial Port 1 be defined by default at the board level, and overridable instead of a main config item?

It could be so. There would need to be an option not just to override it but also to disable it from the configuration too, for cases where users know it won't ever be used.

boelle, 2019-02-20:
@adamnock still having this issue?

adamnock, 2019-02-20:
Nope, that was all sorted. 

My only issue now is my Re-arm blew up because im an idiot.
Probably not a Marlin config flag to fix that

closed 2019-02-20 by adamnock

sl1pkn07, 2019-05-25:
i have the same issue now whit the usb connection over MacOS (the pc more near at printer)

i have configured the unit like @oakdesign and the wiki say (http://marlinfw.org/docs/basics/install_rearm.html)

when open the pronterface and i try connect with the unit. the connection is OK. but any command inserted is recognized as Unknown, include the internal commands launched from pronterface iself 

[image]

 the printer works iself over the LCD

the configuration files

greetings

Bob-the-Kuhn, 2019-05-26:
I tried your two config giles with the latest bugfix_2.0.x and had problems.  

I never saw the symptoms you reported.  What I did see was the PC complaining about an unrecognized USB device.  The USB port appears to come up properly and then die.  

I then switched to a virgin configuration, changed serial port to -1 and change the motherboard to RE_ARM.  That worked.

It appears that you've run acros some combination of config options that is causing problems.

Please open a new issue for this problem.  Please include all the info you've posted here.

syurev, 2019-05-27:
For me uncomment SDSUPPORT with no shield attached caused same issue:
> "What I did see was the PC complaining about an unrecognized USB device. The USB port appears to come up properly and then die."
I`ve not tested it with shield yet...

syurev, 2019-05-27:
> I never saw the symptoms you reported. What I did see was the PC complaining about an unrecognized USB device. The USB port appears to come up properly and then die.
I then switched to a virgin configuration, changed serial port to -1 and change the motherboard to RE_ARM. That worked.

Same issue with Re-Arm with no shield. Trying to compile clear configuration - all(serial port and SD) works fine.
I began to turn on the options one by one, and the board hung again after I changed TEMP_SENSOR_BED from 0 to 1.

sl1pkn07, 2019-05-27:
as note, set TEMP_SENSOR_BED to 0, this bug https://github.com/MarlinFirmware/U8glib-HAL/issues/10 is completly gone

sl1pkn07, 2019-05-27:
@Bob-the-Kuhn @syurev 

https://github.com/MarlinFirmware/Marlin/issues/14152