MarlinFirmware/Marlin #10782: 2.0 BUG? #define FAST_PWM_FAN - not compiling

rockman256, 2018-05-19:
Ramps FD V1
Fan Pin  - 2

sketch\src\module\temperature.cpp: In static member function 'static void Temperature::setPwmFrequency(pin_t, int)':

sketch\src\module\temperature.cpp:1385:36: error: 'digitalPinToTimer' was not declared in this scope

       switch (digitalPinToTimer(pin)) {

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2019629/Marlin.zip)

Bob-the-Kuhn, 2018-05-20:
The short term fix is to disable **FAST_PWM_FAN** in Configuration.h.

I'll need to investigate further on what DUE is capable of and what we've implemented.

Bob-the-Kuhn, 2018-05-20:
FAST_PWM_FAN is only supported by the 8 bit AVR CPUs.  There are no plans to support it on the 32 bit CPUs.

I'll add a sanity check for this.

rockman256, 2018-05-20:
I want to explain the problem of PWM signal and turbo fans.
https://www.aliexpress.com/item/Original-Sunon-EF50151B1-C02C-A99-5015-12V-1-92W-50-50-15mm-Ultra-Quiet-Humidifier-Turbo/32820917468.html
With direct connection to the source, the fan starts smoothly from 4 volts.
But when you connect to the PWM signal - it either does not work, or works at full power at 250.
I tried to use #define FAN_SOFT_PWM - the same result.
So I thought that #define FAST_PWM_FAN will help to solve this problem.

rockman256, 2018-05-20:
Another Chinese turbo fan also works fine, but it's cheap and vibrates strongly.

rockman256, 2018-05-20:
change to 400 Hz - it is optimal worked.

closed 2018-05-20 by rockman256

nonodu74, 2019-08-30:
Hi rockman256,

Which parameter did you change to 900Hz ?

nonodu74, 2019-09-19:
Hi rockman256,

Where do you change frequency to 900Hz ?
Does it affects other components on the board ?

Thks