MarlinFirmware/Marlin #10784: Bug with 3 extruder + switching extruder

mtmello2003, 2018-05-19:
With this scenario: 3 extruder, 2 stepper motor and 1 servo (to move between E0 and E1), after I adjust marlin 2.0 configs, the servo no longer move and the second extruder motor do not respect the steps/mm defined (move really very slow, almost stopped).

I tested same config in 1.1.8 bugfix and got same issue.

Configs:

#define EXTRUDERS 3

// A dual extruder that uses a single stepper motor
#define SWITCHING_EXTRUDER
#if ENABLED(SWITCHING_EXTRUDER)
#define SWITCHING_EXTRUDER_SERVO_NR 1
#define SWITCHING_EXTRUDER_SERVO_ANGLES { 145, 25 } // Angles for E0, E1[, E2, E3]
#if EXTRUDERS > 3
#define SWITCHING_EXTRUDER_E23_SERVO_NR 1
#endif
#endif

PS:
(just two servo angles: E0, E1. E2 do not work with servo, just the stepper motor )

#define DISTINCT_E_FACTORS

#define DEFAULT_AXIS_STEPS_PER_UNIT   {160.886, 160.522, 8151.889, 701*0.25, 701*0.25}
#define DEFAULT_MAX_FEEDRATE          { 300, 300, 3.6, 25, 25 }
#define DEFAULT_MAX_ACCELERATION     { 800, 800, 3.6, 25, 25 }

If I comment like this line:
//#define SWITCHING_EXTRUDER

The T0 motor run ok and the T1 motor run also ok! Then I can confirm that the steps/mm is ok and phisical configuration is also ok.

The problem happen in the behavior of 2 steppers motor and only one with a servo + angles defined...

GMagician, 2018-05-19:
> the servo no longer move

look at #10766

mtmello2003, 2018-05-19:
Applying the change on tool_change.cpp:
#define DO_SWITCH_EXTRUDER true //(SWITCHING_EXTRUDER_SERVO_NR != SWITCHING_NOZZLE_SERVO_NR)

My servo is working when tool changes but the another extruder motor no longer move.
T0 is ok. T1 is ok (servo move to the defined angle and reversing motor is also ok) but T2 does nothing. Motor do not respond.

GMagician, 2018-05-19:
Fix reported in #10776 is temporary, a PR has been done to really solve it.
I have to check if I'm able to solve also other problem, can you post your configuration files?

mtmello2003, 2018-05-19:
[Marlin-bugfix-2.0.x.rar.txt](https://github.com/MarlinFirmware/Marlin/files/2019872/Marlin-bugfix-2.0.x.rar.txt)

My entire marlin config files. Plz rename, removing .txt from file.
Thanks!

thinkyhead, 2018-05-19:
The patch has been merged. Please re-test the latest code, and hopefully all is working!

mtmello2003, 2018-05-20:
I did the download of latest 2.0 bugfix. 

Better now then before, ´cause I just configured both config and config_adv files and T0 and T1 worked ok but T2 still "dead"...

GMagician, 2018-05-20:
Just to be sure...is in your tested code 65270d6 present?

mtmello2003, 2018-05-20:
I did the download of 2.0-bugfix file yesterday but the PR code 65270d6 in not present!

Then, I guess others updates are not included also...

Where can I download the latest bundle, please?

I used this URL http://marlinfw.org/meta/download/

GMagician, 2018-05-20:
OT
@thinkyhead I was looking at EXTRUDER_RUNOUT_PREVENT and it seems that it move only E0...what about EXTRUDERS >2?

GMagician, 2018-05-20:
OT
@thinkyhead in stepper_indirection.h:
```c++
#elif EXTRUDERS > 2
  #define E_STEP_WRITE(v) do{ if (current_block->active_extruder < 2) { E0_STEP_WRITE(v); } else if (current_block->active_extruder < 4) { E1_STEP_WRITE(v); } else { E1_STEP_WRITE(v); } }while(0)
```
extruder < 4 and else is same define...can it be reduced?

thinkyhead, 2018-05-22:
@GMagician — Either it can be reduced, or the last case is supposed to be E2. I'll look at the context and see what makes sense there.

thinkyhead, 2018-05-22:
Yes, that middle clause can go away. I've just pushed the change.

Note that it has no effect on this issue, however. I'll still need to see why T2 is "dead."

By "dead" I assume you mean that the second extruder isn't rotating.

If there's something funky about `EXTRUDER_RUNOUT_PREVENT` I could look into that too. But maybe I'll see about this issue first.

GMagician, 2018-05-22:
> But maybe I'll see about this issue first.

of course, I was trying to figure out what is the cause of this and I saw such code

mtmello2003, 2018-05-22:
By "dead" I assume you mean that the second extruder isn't rotating.

@thinkyhead Exactly. "dead" means that the extruder stepper motor do not move.

thinkyhead, 2018-05-23:
Does it move properly if you turn off the "switching extruder" option and try it as just plain `T1`? I'd like to confirm that the code is generally working for the stepper motor attached to the E1 plug.

mtmello2003, 2018-05-23:
Yes, without using "switching extruder" option T0 and T1 extruder motor move very well.

GMagician, 2018-05-26:
How do you move T1, using lcd? 

@thinkyhead  when SWITCHING_EXTRUDER then 'E_MANUAL' will always be defined as 'EXTRUDERS'...is it correct?

GMagician, 2018-05-26:
@mtmello2003 try to change in 'Conditional_LCD.h' to following code:
```c++
#if ENABLED(SWITCHING_EXTRUDER)                               // One stepper for every two EXTRUDERS
  #if EXTRUDERS > 4
    #define E_STEPPERS    3
    #define E_MANUAL      3
  #elif EXTRUDERS > 2
    #define E_STEPPERS    2
    #define E_MANUAL      2
  #else
    #define E_STEPPERS    1
    #define E_MANUAL        EXTRUDERS
  #endif
```

last E_MANUAL define inside 'else'

thinkyhead, 2018-05-26:
After checking, I see that `E_MANUAL` should in face be equal to `EXTRUDERS`. However, the menu should only show the move option for the currently-selected switched extruder, and should not show the one that isn't selected. So that will need a patch.

thinkyhead, 2018-05-26:
These would come right after the tool-change options. This seems to be the only sensible way.
```cpp
#if ENABLED(SWITCHING_EXTRUDER) || ENABLED(SWITCHING_NOZZLE)

  // Only the current extruder...
  MENU_ITEM(submenu, MSG_MOVE_E, lcd_move_get_e_amount);

  // ...and the non-switching extruder
  #if EXTRUDERS == 5
    MENU_ITEM(submenu, MSG_MOVE_E MSG_MOVE_E5, lcd_move_get_e4_amount);
  #elif EXTRUDERS == 3
    MENU_ITEM(submenu, MSG_MOVE_E MSG_MOVE_E3, lcd_move_get_e2_amount);
  #endif

#else

  MENU_ITEM(submenu, MSG_MOVE_E, lcd_move_get_e_amount);
  #if E_MANUAL > 1
    MENU_ITEM(submenu, MSG_MOVE_E MSG_MOVE_E1, lcd_move_get_e0_amount);
    MENU_ITEM(submenu, MSG_MOVE_E MSG_MOVE_E2, lcd_move_get_e1_amount);
    #if E_MANUAL > 2
      MENU_ITEM(submenu, MSG_MOVE_E MSG_MOVE_E3, lcd_move_get_e2_amount);
      #if E_MANUAL > 3
        MENU_ITEM(submenu, MSG_MOVE_E MSG_MOVE_E4, lcd_move_get_e3_amount);
        #if E_MANUAL > 4
          MENU_ITEM(submenu, MSG_MOVE_E MSG_MOVE_E5, lcd_move_get_e4_amount);
        #endif // E_MANUAL > 4
      #endif // E_MANUAL > 3
    #endif // E_MANUAL > 2
  #endif // E_MANUAL > 1

#endif
```

mtmello2003, 2018-05-26:
> @mtmello2003 try to change in 'Conditional_LCD.h' to following code:

With this modification, desapear from menu the option to move the servo ("select E1/2") , inside MOVE AXIS menu and the second stepper motor still dead in anyway.

thinkyhead, 2018-05-26:
Clearly, something is amiss, but I cannot find any logical error in the handling of the movement of the second extruder stepper for `SWITCHING_EXTRUDER`. As you can see, in `stepper_indirection.h` the `E_STEP_*` macros for that case are totally correct...
```cpp
#if ENABLED(SWITCHING_EXTRUDER)
  #if EXTRUDERS > 4
    . . .
  #elif EXTRUDERS > 3
    . . .
  #elif EXTRUDERS > 2
    #define E_STEP_WRITE(v) do{ if (current_block->active_extruder < 2) { E0_STEP_WRITE(v); } else { E1_STEP_WRITE(v); } }while(0)
    #define NORM_E_DIR() do{ switch (current_block->active_extruder) { case 0: E0_DIR_WRITE(!INVERT_E0_DIR); break; case 1: E0_DIR_WRITE( INVERT_E0_DIR); break; case 2: E1_DIR_WRITE(!INVERT_E1_DIR); } }while(0)
    #define REV_E_DIR()  do{ switch (current_block->active_extruder) { case 0: E0_DIR_WRITE( INVERT_E0_DIR); break; case 1: E0_DIR_WRITE(!INVERT_E0_DIR); break; case 2: E1_DIR_WRITE( INVERT_E1_DIR); } }while(0)
  #else
    . . .
  #endif
```
So the error must lie elsewhere, and we'll keep looking.

mtmello2003, 2018-05-27:
I did it right now and could see that the MOVE AXIS menu is really more reasonable for the servo + extruder environment...

unfortunately, T2 still dead with LA disabled and with LA enable, I couldn´t compile:

sketch\src\HAL\HAL_LPC1768\include\digipot_mcp4451_I2C_routines.c:27:0: error: expected '=', ',', ';', 'asm' or '__attribute__' at end of input
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

thinkyhead, 2018-05-27:
I can see no issues with `digipot_mcp4451_I2C_routines.c` or the flagged `#include` line. I am able to compile the latest `bugfix-2.0.x` for `BOARD_MKS_SBASE` successfully with `LIN_ADVANCE` enabled. If you still cannot compile, then please ZIP up your most up-to-date `Configuration.h` and `Configuration_adv.h` files and post them in a reply for further testing.

thinkyhead, 2018-05-27:
Looking at the last configs you posted, apart from their being pretty mangled, you also have `SINGLENOZZLE` enabled. If that option is still enabled, you should disable it.

thinkyhead, 2018-05-27:
Here are some shiny new configs for the latest `bugfix-2.0.x` branch with the mangled comment lines and indentation replaced with non-mangled and properly indented lines. These have all your options just as you configured them, except that `SINGLENOZZLE` is disabled.

[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/2042268/Configuration_adv.h.txt)
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/2042269/Configuration.h.txt)

mtmello2003, 2018-05-27:
> I've examined the last configs you posted, and apart from their being pretty mangled, you also have SINGLENOZZLE enabled. If that option is still enabled, you should disable it!

> That should fix your T2 issue.

OMG! For me, using SINGLENOZZLE makes all sense ´cause that´s my scenario (using E3DV6 to receive all extruders) but when I disabled that tag, I  got error from SanityCheck that demands me to define TEMP_SENSOR_1 and _2 (config.h). 

Did it and started to get error about missing TEMP_1_PIN , TEMP_2_PIN, HEATER_1_PIN, HEATER_2_PIN.

I add all missing _PIN on pin_RAMPS.h and could compile:

Unfurtunately:
T0 ok
T1 ok
T2 still dead

Just to double check, I disabled the SWITCHING EXTRUDER tag and did the test again and my second stepper moved OK...

thinkyhead, 2018-05-27:
I'll make sure the temp sensor sanity check makes more sense for switching extruder.

thinkyhead, 2018-05-27:
A single hotend with a switching extruder plus an additional E-stepper is potentially supportable, but I'll have to check the code to whether `SINGLENOZZLE` has other expectations or side-effects.

Interesting to hear that even with `SINGLENOZZLE` disabled your second stepper motor still isn't engaging. We'll continue to search for the anomaly.

boelle, 2019-02-20:
@mtmello2003 is this still an issue?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead

Reduce395, 2020-04-30:
@thinkyhead I was just setting up a config similar to this (having a switching extruder configured with a direct drive hotend) and I'm still getting the same issues as described by mtmello. 

2 extruders works perfectly fine; 3 and up will not work.
With singlenozzle also enabled, the second stepper will move, but very slowly. I tried running it at different speeds (G1 E10 F10000) and it continues to run at the slow pace.
With singlenozzle disabled, marlin throws an error about the number of temp sensors/hotends. Circumventing that, the second stepper continues to fail and won't move at all.
I'm on Marlin-bugfix-2.0.x for the SKR 1.3. The motor is tested as working (with switching disabled or choosing 2 extruders, the motor runs perfectly fine).

My original plan was to configure the switching mechanism to switch to X angle so the switching extruder could feed one of the filaments. The other extruder (direct drive extruder) would also be moving at the same rate (they're different motors with different esteps, titan aero has a 3:1 gear ratio, so I can't just directly mirror them- plus I would have to somehow read when the switching motor inverts) to feed in the filament all the way down. Once the filament was purged, I would then switch to Y angle so the left filament path is now untensioned and the direct drive extruder would take it from there.
Is this a possible configuration?

Thanks!