MarlinFirmware/Marlin #10785: Unsupported display, is there a way to make it supported ?

AndKe, 2018-05-19:
The display with module looks just like "REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER" , with buzzer, and rotary switch.
- except, only one EXT connector is installed, the other *could be* used for SD card slot (which is not installed on the card)  - it has a 20pin LCD connector, with the two last pins being used for backlit.

The pinout on EXT1 (the only one)  is: 
1: +5v   , positive side of the buzzer, backlit
2: GND , used for misc devices/switches.
3: RW (5th pin of LCD module)
4: RS (4th pin of LCD module)
5: E (6th pin of LCD module)
6: rotary switch rotation data
7: switch two (reset ?) - not used
8: rotary switch rotation data
9 rotary switch button
10: via resistor to gate of Q1,  emitter of Q1 is connected to GND, collector to buzzer(negative)  - so this pin drives the beeper.

is if feasible to make the display, buzzer, rotary switch work on EXT1 ?  (it's mechanically compatible, fits right in, the question is whatever it is a major task to support that hardware) 
- also , I am surprised that no more of the displays pins are in use.

AndKe, 2018-05-19:
I could give it a shot by modifying pins_ANET_10.h  , but one of my problems now is the correlation between EXT1 pin 5 and pin number in "defines"  , aka, I need to know the firmware pin number of all EXT1 pins..

AndKe, 2018-05-20:
The controller is an CR-10S upgrade (apparently fully RAMPS_14_EFB compatible)
https://www.ebay.com/itm/Creality-Upgrade-CR-10S-S4-S5-3D-Printer-Motherboard-Circuit-Control-Board-B2Q8/183101852072?hash=item2aa1b889a8:g:QXUAAOSwL7ValXTX

The display comes with A10 printers: (basically a CR-10 clone)

  https://www.ebay.com/sch/i.html?_from=R40&_trksid=p2047675.m570.l1313.TR0.TRC0.H0.XA10+3d+printer.TRS0&_nkw=A10+3d+printer&_sacat=0

- the A10 printer works really well, good precision, but I am replacing the controller for a proper 2560 and current Marlin.  - expect others to do that as well :)

Bob-the-Kuhn, 2018-05-20:
These are the pin names you'll need to use, assuming it's a clone/derivative of the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

```
Marlin name       The pinout on EXT1 (the only one) is:
                  1:  +5v , positive side of the buzzer, backlit
                  2:  GND , used for misc devices/switches.
LCD_PINS_D4       3:  RW (5th pin of LCD module)
LCD_PINS_RS       4:  RS (4th pin of LCD module)
LCD_PINS_ENABLE   5:  E (6th pin of LCD module)
BTN_EN1           6:  rotary switch rotation data
KILL_PIN          7:  switch two (reset ?) - not used
BTN_EN2           8:  rotary switch rotation data
BTN_ENC           9:  rotary switch button
BEEPER_PIN        10: via resistor to gate of Q1, emitter of Q1 
```
 
If the encoder direction is reversed then switch BTN_EN1 & BTN_EN2.

None of these signals are special.  they can be assigned to any pin except SCK, MOSI & MISO.  Those are usually reserved for the SD card.

If you don't know the logical pin numbers on the controller connector you can use the M43 commands to figure that out.  M43 is enabled by the PINS_DEBUGGING flag in Configuration_adv.h

AndKe, 2018-05-20:
M43 is a great tool.
I should have it working now, but something isn't 

M43 shows that

AndKe, 2018-05-20:
I've measured everything end-to-end, If I were able to define pins correctly - it should have worked.
those defines, if they have been actually compiled, would work:
I find the "LCD_PINS_D4" name to be a strange label for RW on ST7920 , assuming it's a cosmetic issue only.

    #define BEEPER_PIN       37
    #define LCD_PINS_RS      27   //"RS" on display
    #define LCD_PINS_ENABLE  25  // "E" on display
    #define LCD_PINS_D4      29  // "R/W" on display
    #define BTN_EN1          23
    #define BTN_EN2          17
    #define BTN_ENC          35

I've tried to cram them in the beginning of Configuration.h without success

thinkyhead, 2018-05-20:
> I've tried to cram them in the beginning of `Configuration.h` without success

It's cleaner if you put them into the pins file for your board. The compiler will complain if they are given values in config and then the pins file tries to define them again with different values.

Bob-the-Kuhn, 2018-05-21:
Do you have the logical pin numbering for the controller side?

A pin diagram will do.  A schematic is even better.

Without those then you can use the **M43 W I** and **M43 T I** commands to figure out the logical pin mapping.

I'd start with **M43 W I **.  This command just watches the pins and reports any changes of state.  What I do is plug one end of a jumper into ground and then touch the unknown pins with the other.  75% of the time the display will report the logical pin number and the signals assigned to it.

The **M43 T I** command toggles pins.  I attach an LED to the unknown pin and then start the command toggling pins.  When I see the LED flash I look at the display and note the number.  Usually I'm too slow to catch the correct number so I'll re-start the test over a much smaller range at a slower rate.

 ```
  *  M43 W       - Watch pins -reporting changes- until reset, click, or M108.
   *                  P<pin>  Pin to read or watch. If omitted, read/watch all pins.
   *                  I       Flag to ignore Marlin's pin protection.

   *
   *  M43 T       - Toggle pin(s) and report which pin is being toggled
   *                  S<pin>  - Start Pin number.   If not given, will default to 0
   *                  L<pin>  - End Pin number.   If not given, will default to last pin defined for this board
   *                  I<bool> - Flag to ignore Marlin's pin protection.   Use with caution!!!!
   *                  R       - Repeat pulses on each pin this number of times before continueing to next pin
   *                  W       - Wait time (in miliseconds) between pulses.  If not given will default to 500
```

AndKe, 2018-05-21:
After much narrowing-down by looking at other displays, I realized that this display is already supported as CR10_STOCKDISPLAY
:)

closed 2018-05-21 by AndKe