MarlinFirmware/Marlin #10786: [2.0] Some changes and requested additions to common i2c LCDs and backpacks

AletheianAlex, 2018-05-19:
Since there is an upcoming display cleanup/modularization, I thought I’d throw this in: I recently set up a few i2c IO expander LCD boards which are not currently implemented, so I thought I would leave the information here in case it is desirable to include since these are two common and cheap boards (waveshare--or similar--PCF8574 IO expander breakout, and FC-113 i2d led backpack which is similar to the SAINSMART_I2C backpacks, but slightly improved).   

(The easiest method was using the sainsmart define in configuration.h and redefining LCD size and i2c address, so it may be sufficient to just add a note that it works with these backpack/expanders, and a note on how to exclude the old lib from the .ini build env.)

--Also, SAINSMART_I2C LCDs and backpacks are (randomly) using either pcf8574T, 8574A, or 8574AT, in addition to the 8575 they used in the past, which could cause confusion since it makes the current i2c address incorrect in conditionals_LCD as well as several included libraries: defaults for pcf8574T = 0x27, AND 8574A or 8574AT = 0x3F, and the waveshare board comes default with all jumpers on, which sets it at 0x38 (pulling all 3 jumpers = 0x3F)

--All models are working with the fmalpartida/new-liquidcrystal/ library

--PlatformIO env settings will not work unless LiquidCrystal@1.3.4 is added to the envelope's lib_deps, and LiquidCrystal_I2C is added to the envelope’s lib_ignore 

--Either of these works in Configuration.h but require some changes in the /Marlin/src/lcd/ ultralcd HD44780 headers and/or /Marlin/src/inc Conditionals_LCD.h:

```
#define LCD_SAINSMART_I2C_1602

#define LCD_SAINSMART_I2C_2004

#define LCM1602
```

Current changes to get these boards functional (could also redefine in configuration.h if working off the sainsmart setup):
In Conditionals_LCD.h : line 208 address is coded in Sainsmart macro:

```C++
#if ENABLED(LCD_SAINSMART_I2C_1602) || ENABLED(LCD_SAINSMART_I2C_2004)

  #define LCD_I2C_TYPE_PCF8575
  #define LCD_I2C_ADDRESS 0x27 // <===== change this to I2C Address of the port expander

  #if ENABLED(LCD_SAINSMART_I2C_2004)
    #define LCD_WIDTH 20
    #define LCD_HEIGHT 4
  #endif
```

OR for the generic LCM1602 ULTRALCD_IMPL_HD44780_H

```C++
#elif ENABLED(LCM1602)
  LCD_CLASS lcd(0x27, 2, 1, 0, 4, 5, 6, 7, 3, POSITIVE); // <===== change this to I2C Address of the port expander
```

NOTES about LCD_CLASS:

In ULTRALCD_COMMON_HD44780_H : line 130, these match up fine with the sainsmart, FC-113, and Waveshare boards:

```C++
// Create LCD class instance and chipset-specific information
#if ENABLED(LCD_I2C_TYPE_PCF8575)
  // note: these are register mapped pins on the PCF8575 controller not Arduino pins
  #define LCD_I2C_PIN_BL  3
  #define LCD_I2C_PIN_EN  2
  #define LCD_I2C_PIN_RW  1
  #define LCD_I2C_PIN_RS  0
  #define LCD_I2C_PIN_D4  4
  #define LCD_I2C_PIN_D5  5
  #define LCD_I2C_PIN_D6  6
  #define LCD_I2C_PIN_D7  7
```

Also this works ULTRALCD_IMPL_HD44780_H

This LCD_CLASS structure works with ehe newliquidcrystal lib  (as long as backlight pin assignment and polarity match):

```C++
#elif ENABLED(LCM1602)
  LCD_CLASS lcd(0x27, 2, 1, 0, 4, 5, 6, 7, 3, POSITIVE);
```

The is a possible third option that I have not tried using the old liquid crystal_i2c lib which was not working for me:

There is a generic LCD_I2C_TYPE_PCA8574 type that is defined if RA_CONTROL_PANEL is enabled, which assumes a PCF8574T with jumpers opened, but the lcd class struct LCD_CLASS lcd(LCD_I2C_ADDRESS, LCD_WIDTH, LCD_HEIGHT) is incompatible with the libs using a different pinout assumption or method... unless the rest of the parameters are defined up for each setup case and  passed in the init process as in the RA_CONTROL_PANEL? As in ultralcd_common_HD44780.h : line 322 (inside static void lcd_implementation_init). Regardless, it did not seem to work with the 2 boards I was seeing up in its current form.  

I don’t know what the cleanest option is. Definitions can be in configuration.h, but different libs mean changes in the platform.ini envelopes, as well as different variables to pass to the different libs' LCD_CLASS class structures.

Here are photos of the chip variants' default and jumper addresses, as well as the two boards I used. 

[image]
[image]

thinkyhead, 2018-05-20:
I will start with these basic changes to neaten up the LCD code a bit, since these are all related.

#### Conditionals_LCD.h
```cpp
  #elif ENABLED(LCM1602)

    #define LCD_I2C_TYPE_PCF8575
    #define LCD_I2C_ADDRESS 0x27   // I2C Address of the port expander
    #define ULTRA_LCD

```

#### ultralcd_impl_HD44780.h
```cpp
#if ENABLED(LCD_I2C_TYPE_PCF8575)
  // NOTE: These are register-mapped pins on the PCF8575 controller, not Arduino pins.
  #define LCD_I2C_PIN_BL  3
  #define LCD_I2C_PIN_EN  2
  #define LCD_I2C_PIN_RW  1
  #define LCD_I2C_PIN_RS  0
  #define LCD_I2C_PIN_D4  4
  #define LCD_I2C_PIN_D5  5
  #define LCD_I2C_PIN_D6  6
  #define LCD_I2C_PIN_D7  7

  #include <Wire.h>
  #include <LCD.h>
  #include <LiquidCrystal_I2C.h>
  #define LCD_CLASS LiquidCrystal_I2C

  LCD_CLASS lcd(LCD_I2C_ADDRESS, LCD_I2C_PIN_EN, LCD_I2C_PIN_RW, LCD_I2C_PIN_RS, LCD_I2C_PIN_D4, LCD_I2C_PIN_D5, LCD_I2C_PIN_D6, LCD_I2C_PIN_D7
    #if ENABLED(LCM1602)
      , LCD_I2C_PIN_BL, POSITIVE
    #endif
  );

#elif ENABLED(LCD_I2C_TYPE_MCP23017)
 . . .
```

> …but the lcd class struct `LCD_CLASS lcd(LCD_I2C_ADDRESS, LCD_WIDTH, LCD_HEIGHT)` is incompatible with the libs…

We can add as many custom LCD initializations as are needed to deal with differences. We just need to know what the initializers ought to be for all the common controllers.

> Here are photos of the chip variants' default and jumper addresses, as well as the two boards I used.

We can add an option to override the default I2C address 0x27 next to the LCD options in `Configuration.h`, or just instruct them to edit the `Conditionals_LCD.h` file — for savvy users who are doing their own solder jumpers.

thinkyhead, 2018-05-20:
I'm confused concerning the LCM1602 and how it's currently configured and initialized in Marlin. It does appear just like a PCF8575 interface, since all the pin numbers match! But when I try to incorporate the changes outlined above I get some strange errors back from Travis CI…
- "Can't find LCD.h" 
- "`POSITIVE` is not defined."
However, if I leave the code alone then the compiler has no problem with…
```cpp
LCD_CLASS lcd(0x27, 2, 1, 0, 4, 5, 6, 7, 3, POSITIVE);
```
It's pretty weird.

AletheianAlex, 2018-05-20:
Understood, I can confirm that with the HD44780 header changes, it won't build, BUUUT! with just the conditionals_lcd.h changes added, and the new library, it builds and functions with either the LCM1602 or Sainsmart options defined, which it didn't before.  The LCM1602 stuff is spread across a few files, so maybe there was a change to include the class from a different library?  

The library (LiquidCrystal_i2c_1.1.2) for the older YwRobot backpack only took address, height/columns/rows as arguments, and I could not get it working with the backpacks/expanders I have here in 2.0/32-bit in Marlin or alone with a 'hello world' sketch in STM32duino/Arduino IDE, which is why I turned to the newliquidcrystal 1.3.4 lib at https://bitbucket.org/fmalpartida/new-liquidcrystal/wiki/Home which also worked with the same code on an Uno. Arduino IDE also classifies the LiquidCrystal_i2c_1.1.2 lib examples as incompatible with 32-bit.  IF AVR code works with the new LiquidCrystal@1.3.4 and IF the classes structures required by Marlin are represented in it, maybe drop LiquidCrystal_i2c_1.1.2?

Here are the setups that worked:

set in the .ini env:

`lib_deps = LiquidCrystal@1.3.4 `

and 

`lib_ignore = LiquidCrystal_I2C` 

In configuration.h

```C++
#define LCM1602
#define LCD_WIDTH 20
#define LCD_HEIGHT 4
#define LCD_I2C_ADDRESS 0x3F
```
OR 

```C++
#define LCD_SAINSMART_I2C_2004
#define LCD_I2C_ADDRESS 0x3F 
```

Both work.  I tried defining the address to something bonkers to make sure the define is working and it produced nothing on the display, so that confirmed it.

closed 2019-07-11 by shitcreek