MarlinFirmware/Marlin #10792: Prints getting Random Y Step skips- Marlin-bugfix-1.1.x

bentech4u, 2018-05-20:
Hi All,

I am net to printing and last one year i have been trying to build my own printer.My electronics is RAMPS 1.5 + Ardunio + A4988 drivers. My firmware is **Marlin-bugfix-1.1.x**

the printer was working fine and from last one week i am getting random step skips on Y axis like below:

[https://s9.postimg.cc/6ydnp800v/IMG_20180520_062511.jpg](url)
[https://s9.postimg.cc/4h1wi2qfz/IMG_20180520_095447.jpg](url)
[https://s9.postimg.cc/f3vpnjw1b/IMG_20180520_095501.jpg](url)

I am not sure this is because of my values in firmware

`#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 4000, 94.69 }

#define DEFAULT_MAX_FEEDRATE          { 150, 150, 3, 25 }

#define DEFAULT_MAX_ACCELERATION      { 2000, 2000, 100, 10000 }

#define DEFAULT_ACCELERATION          2000    // X, Y, Z and E acceleration for printing moves
#define DEFAULT_RETRACT_ACCELERATION  2000    // E acceleration for retracts
#define DEFAULT_TRAVEL_ACCELERATION   2000    // X, Y, Z acceleration for travel (non printing) moves

#define DEFAULT_XJERK                 15.0
#define DEFAULT_YJERK                 15.0
#define DEFAULT_ZJERK                  0.4
#define DEFAULT_EJERK                  5.0`

PLease help me to fix this issue

c0un7-z3r0, 2018-05-20:
@bentech4u  Could it be related to #10446 ?

bentech4u, 2018-05-20:
Yes most of my prints like this and it's random 

so it's from firmware? if so which is stable version

c0un7-z3r0, 2018-05-20:
I guess the stable release is the https://github.com/MarlinFirmware/Marlin/releases/tag/1.1.8

thinkyhead, 2018-05-20:
Duplicate of #10446

closed 2018-05-20 by thinkyhead