MarlinFirmware/Marlin #10793: Where is UNKNOWN_Z_NO_RAISE in the latest update?

Phaelz, 2018-05-20:
I've just updated the marlin to correct the motion problem when Segmented move was on with mesh bed leveling.

Comparing the two versions, now I notice my bed crashing at the bottom (it was already on MAX_Z_POS) when I home it for the first time.

There was this line of code on Configuration.h that prevented that from happening.

It was UNKNOWN_Z_NO_RAISE. It is not there anymore. Nowhere on the new marlin.

Am I missing it? It changed names?

Phaelz, 2018-05-20:
Whoa... I really need to sleep. Sorry, my bad. BIG TIME mistake here. I downloaded version 1.0 of Marlin thinking it was the latest bugfix. I ran 2 hours of experiments here, it was all messed up hahahahhaha

Please close this thread, I am going to erase the comments I made in the last 30 min.

Thanks, @GMagician .

closed 2018-05-20 by Phaelz

GMagician, 2018-05-20:
lol

Phaelz, 2018-05-20:
The funny thing is everywhere on the code I looked I knew that something was wrong. It was all different. Then I started thinking about someone pulling a prank here.. I'm outta here, sleep deprivation is serious stuff hahaha

thinkyhead, 2018-05-20:
Damn that Loki…