MarlinFirmware/Marlin #10794: 2.0 - Bltouch & PID - PWM  interference

rockman256, 2018-05-20:
The problem is when the PID hotent is running - BLTOUCH **spontaneously starts to throw the probe outward**, basically it happens at the beginning of the printing and the probe touches the skirt. I'm afraid that one day it will damage the probe.
My initial code is:
G21; set units to millimeters
G90; use absolute positioning
M140 S [bed0_temperature] T0; set bed temp
M190 S [bed0_temperature] T0; wait for bed temp
G1 F1000 Z15
G28 W; home all without mesh bed level
G29; mesh bed leveling
M420 Z10
M82; absolute extrusion mode
M104 S [extruder0_temperature] T0; set extruder temp
M109 S [extruder0_temperature] T0; wait for extruder temp
G92 E0.0; reset extruder distance position
I tried to include the function - **#define DEACTIVATE_SERVOS_AFTER_MOVE**, but I did not get a result, so I understand that this function does not work for BLTouch?
Interference goes exactly on the signal wire, I tested, if during operation hotend switch bltouch to another port servo - the probe does not go out somatically outside.
**Is it possible to disable the servo signal after G29?**

GMagician, 2018-05-20:
I think this is due to issues on 5V, GND level or EMI. I had your issues when bed PWM was working (solved with flyback diodes) and now I have it when moving servo (a new one since old was damaged but used to work with no issues)

rockman256, 2018-05-20:
On power with arduino (ramps FDv1) there is no problem, I also tried to power up from a separate power supply, had the same effect. The interference goes only through the signal cable. Of course, you can solve the problem by soldering the PWM filter. But I think it's much easier to just turn off the signal, at the moment when the servo is not needed.

GMagician, 2018-05-20:
Turning off servo, as I know you have 5 pins on bl-touch. 2 GND, 1 5V, 1 signal and 1 PWM.
There is no "enable" or something else, DEACTIVATE_SERVOS_AFTER_MOVE will stop timer and, not sure, stop pwm so what else can be done?

GMagician, 2018-05-20:
> On power with arduino (ramps FDv1) there is no problem,

also boards layout can let them to be prone to noise capturing

rockman256, 2018-05-20:
Repeat - When hotend is on? and i switch bltouch to another port servo (gnd,5v, signal) (port not used at firmware)- the probe does not go out somatically outside.
So signal on signal cable  activated by firmware.

GMagician, 2018-05-20:
> if during operation hotend switch bltouch to another port servo - the probe does not go out somatically outside.

sorry, missed this part

rockman256, 2018-05-21:
11:03:30.025 : PIN: 00        <unused/unknown>                       Input  = 1
11:03:30.030 : PIN: 01        <unused/unknown>                       Input  = 1
11:03:30.030 : PIN: 02        FAN_PIN                                protected
11:03:30.045 : PIN: 03        SERVO3_PIN                             Input  = 1
11:03:30.045 : PIN: 04        SDSS                                   Output = 1
11:03:30.055 : .              SS_PIN                                 Output = 1
11:03:30.055 : PIN: 05        SERVO2_PIN                             Input  = 1
11:03:30.065 : PIN: 06        SERVO1_PIN                             Input  = 1
11:03:30.065 : PIN: 07        SERVO0_PIN                             Output = 0
11:03:30.075 : PIN: 08        HEATER_BED_PIN                         protected
11:03:30.075 : PIN: 09        <unused/unknown>                       Input  = 1
11:03:30.116 : PIN: 10        HEATER_0_PIN                           protected
11:03:30.116 : PIN: 11        CASE_LIGHT_PIN                         Input  = 1
11:03:30.116 : PIN: 12        E0_AUTO_FAN_PIN                        protected
11:03:30.116 : PIN: 13        LED_PIN                                Input  = 0
11:03:30.116 : PIN: 14        <unused/unknown>                       Input  = 1
11:03:30.116 : PIN: 15        <unused/unknown>                       Input  = 1
11:03:30.136 : PIN: 16        LCD_PINS_RS                            Output = 1
11:03:30.136 : PIN: 17        LCD_PINS_ENABLE                        Output = 0
11:03:30.136 : PIN: 18        <unused/unknown>                       Input  = 1
11:03:30.136 : PIN: 19        <unused/unknown>                       Input  = 1
11:03:30.146 : PIN: 20        <unused/unknown>                       Input  = 1
11:03:30.146 : PIN: 21        <unused/unknown>                       Input  = 1
11:03:30.151 : PIN: 22        X_MIN_PIN                              protected
11:03:30.151 : PIN: 23        LCD_PINS_D4                            Output = 0
11:03:30.166 : PIN: 24        Y_MIN_PIN                              protected
11:03:30.166 : PIN: 25        LCD_PINS_D5                            Output = 0
11:03:30.176 : PIN: 26        Z_MIN_PIN                              protected
11:03:30.176 : PIN: 27        LCD_PINS_D6                            Output = 0
11:03:30.186 : PIN: 28        E0_DIR_PIN                             protected
11:03:30.186 : PIN: 29        LCD_PINS_D7                            Output = 0
11:03:30.196 : PIN: 30        <unused/unknown>                       Input  = 1
11:03:30.197 : PIN: 31        BTN_EN2                                Input  = 1
11:03:30.236 : PIN: 32        E2_STEP_PIN                            Output = 0
11:03:30.236 : PIN: 33        BTN_EN1                                Input  = 1
11:03:30.236 : PIN: 34        <unused/unknown>                       Input  = 1
11:03:30.236 : PIN: 35        BTN_ENC                                Input  = 1
11:03:30.236 : PIN: 36        E0_STEP_PIN                            protected
11:03:30.236 : PIN: 37        BEEPER_PIN                             Output = 0
11:03:30.246 : PIN: 38        <unused/unknown>                       Input  = 1
11:03:30.246 : PIN: 39        E1_ENABLE_PIN                          protected
11:03:30.251 : .              Z2_ENABLE_PIN                          protected
11:03:30.251 : PIN: 40        <unused/unknown>                       Input  = 0
11:03:30.261 : PIN: 41        E1_DIR_PIN                             protected
11:03:30.261 : .              Z2_DIR_PIN                             protected
11:03:30.272 : PIN: 42        E0_ENABLE_PIN                          protected
11:03:30.272 : PIN: 43        E1_STEP_PIN                            protected
11:03:30.285 : .              Z2_STEP_PIN                            protected
11:03:30.286 : PIN: 44        Z_ENABLE_PIN                           protected
11:03:30.297 : PIN: 45        E2_ENABLE_PIN                          Output = 0
11:03:30.298 : PIN: 46        Y_ENABLE_PIN                           protected
11:03:30.306 : PIN: 47        E2_DIR_PIN                             Output = 0
11:03:30.307 : PIN: 48        X_ENABLE_PIN                           protected
11:03:30.317 : PIN: 49        MAX6675_SS                             Input  = 0
11:03:30.319 : .              SD_DETECT_PIN                          Input  = 0
11:03:30.333 : PIN: 50        <unused/unknown>                       Input  = 0
11:03:30.334 : PIN: 51        <unused/unknown>                       Input  = 1
11:03:30.348 : PIN: 52        <unused/unknown>                       Input  = 0
11:03:30.348 : PIN: 53        <unused/unknown>                       Input  = 1
11:03:30.353 : PIN: 54 (A 0)  TEMP_1_PIN                             Input  = 1
11:03:30.354 : PIN: 55 (A 1)  TEMP_0_PIN                             protected
11:03:30.363 : PIN: 56 (A 2)  TEMP_BED_PIN                           protected
11:03:30.364 : PIN: 57 (A 3)  TEMP_2_PIN                             Input  = 1
11:03:30.375 : PIN: 58 (A 4)  Z_CS_PIN                               protected
11:03:30.384 : PIN: 59 (A 5)  E2_CS_PIN                              Input  = 1
11:03:30.385 : PIN: 60 (A 6)  Y_CS_PIN                               protected
11:03:30.396 : PIN: 61 (A 7)  E1_CS_PIN                              protected
11:03:30.396 : .              Z2_CS_PIN                              protected
11:03:30.404 : PIN: 62 (A 8)  X_DIR_PIN                              protected
[... cut ...]
11:03:30.461 : PIN: 72        <unused/unknown>                       Input  = 1
11:03:30.472 : PIN: 73        <unused/unknown>                       Input  = 1
11:03:30.472 : PIN: 74        MISO_PIN                               Input  = 0
11:03:30.480 : PIN: 75        MOSI_PIN                               Input  = 1
11:03:30.484 : PIN: 76        SCK_PIN                                Input  = 0
11:03:30.488 : PIN: 77        <unused/unknown>                       Input  = 1
11:03:30.488 : PIN: 78        <unused/unknown>                       Input  = 1

thinkyhead, 2018-05-24:
Try setting `SERVO0_PIN` to 6 and then connecting the BLTouch to the second servo port. Set `SERVO1_PIN` to -1 to keep it out of the way.

rockman256, 2018-05-26:
> 
Try setting SERVO0_PIN to 6 and then connecting the BLTouch to the second servo port. Set SERVO1_PIN to -1 to keep it out of the way.
> 
Tried... And nothing change. 
After hotend is come to stable temperatrure, pid work (10sec - 2min ) - Z-probe is still go out. 
 
10:53:14.511 : N14 M43*33
10:53:14.554 : PIN: 00        <unused/unknown>                       Input  = 1
10:53:14.572 : PIN: 01        <unused/unknown>                       Input  = 1
10:53:14.572 : PIN: 02        FAN_PIN                                protected
10:53:14.572 : PIN: 03        SERVO3_PIN                             Input  = 1
10:53:14.572 : PIN: 04        SDSS                                   Output = 1
10:53:14.578 : .              SS_PIN                                 Output = 1
10:53:14.578 : PIN: 05        SERVO2_PIN                             Input  = 1
10:53:14.592 : PIN: 06        SERVO0_PIN                             Output = 0
10:53:14.592 : PIN: 07        <unused/unknown>                       Input  = 1

thinkyhead, 2018-05-27:
It sounds like there's some kind of signal voodoo happening on the board.
I think we're back to what @GMagician was saying.

boelle, 2019-02-20:
@rockman256 is this still an issue?

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-03-14 by thinkyhead