MarlinFirmware/Marlin #10795: 2.0 BUG? #define POWER_LOSS_RECOVERY - not work correctly

rockman256, 2018-05-20:
#define POWER_LOSS_RECOVERY does not work.
After turning the power off and on.
- resume print
The printer warms up the table and hotend, then XY goes home, returns to the print place.
The display shows "Continuation" and everything is on it.
Further printing does not continue.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2020254/Marlin.zip)

thinkyhead, 2018-05-20:
Yes, there may still be bugs in this new feature. I plan to work on it soon. We have other issues open about it, so I'll just close this one.

closed 2018-05-20 by thinkyhead

ShaggyDog18, 2018-12-26:
similar here:  however, The printer warms up the table and hotend, then XY goes home and... starts printing there, without going to the last printing position...