MarlinFirmware/Marlin #10799: Auto leveling issue - Error: Probing failed

syedrizvi0512, 2018-05-20:
### Description
I am trying to level Flsun Delta Mini using G29 (and other codes) but the probe stops roughly 1.25" above the bed. This I assume is happening because the Z 0 is at that point. Being a beginner at 3D printing and this being my first 3D printer I don't know how to change the offset.

I tried to change Z offset in Slic3r with a -25.40 to see if that bring the nozzle any closer to the bed but hasn't worked. I also changed `#define Z_PROBE_OFFSET_FROM_EXTRUDER 0.25` to a 0.35 and didn't really notice any change. I didn't want to make any further changes.

What should I do now?

Following is the Serial Monitor log from Marlin during this particular error.

`start
echo:Marlin 1.1.8

echo: Last Updated: 2017-12-25 12:00 | Author: (Michael Henke, flsun Kossel Mini)
echo:Compiled: May 20 2018
echo: Free Memory: 3339  PlannerBufferBytes: 1264
echo:V47 stored settings retrieved (902 bytes; crc 9168)
echo:  G21    ; Units in mm
echo:  M149 C ; Units in Celsius

echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X100.00 Y100.00 Z100.00 E90.00
echo:Maximum feedrates (units/s):
echo:  M203 X200.00 Y200.00 Z200.00 E200.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X4000 Y4000 Z4000 E4000
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P2500.00 R3000.00 T3000.00
echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z10.00 E5.00
echo:Auto Bed Leveling:
echo:  M420 S0
echo:Endstop adjustment:
echo:  M666 X0.00 Y0.00 Z0.00
echo:Delta settings: L<diagonal_rod> R<radius> H<height> S<segments_per_s> B<calibration radius> XYZ<tower angle corrections>
echo:  M665 L218.00 R101.00 H280.00 S160.00 B75.00 X0.00 Y0.00 Z0.00
echo:Material heatup parameters:
echo:  M145 S0 H180 B70 F0
echo:  M145 S1 H240 B100 F0
echo:PID settings:
echo:  M301 P24.77 I1.84 D83.61
echo:Z-Probe Offset (mm):
echo:  M851 Z0.25
echo:SD card ok
ok
ok
echo:Home XYZ first
ok
X:0.00 Y:0.00 Z:280.00 E:0.00 Count X:47319 Y:47319 Z:47319
ok
echo:busy: processing
echo:busy: processing
Error:Probing failed
echo:busy: processing
X:-36.00 Y:-70.00 Z:50.00 E:0.00 Count X:26094 Y:22860 Z:18033
ok`

GMagician, 2018-05-20:
> M851 Z0.25

it seems you still have old value. After flashing new firmware I think you have to execute M502 followed by M500 to activate new values.
Otherwise, before homing (G28) execute a M851 with your new value (but after power off you lose it)

About G29 I think that there are some limits over Z position 0, just to be sure not to break into bed.
Limit is Z_PROBE_LOW_POINT in configuration.h

syedrizvi0512, 2018-05-21:
GMagician,
Thanks for the suggestion, I will definitely try it. 
However, since I'm new to this I just want to make sure some of the terminology
> After flashing new firmware

Meaning after I have added the new values?

thinkyhead, 2018-05-21:
> After flashing new firmware

After you've done an "upload" to the board. ("Flash" = "Upload")

GMagician, 2018-05-21:
When you change something in your source files (eg configuration.h) you have to compile and write your compiled sources to the CPU board (flashing/upload).
Boards (mega2560) has a eeprom where some values are stored. 
These values are kept even when you "program" your new code. Hence when you only change values in configuration.h and send them to your board, you haven't change anything but only "default" values.
When you turn on printer, eeprom stored values are read and copied to memory. Values in memory are then used.
When you execute M502 you just tell system to copy "default" values to memory (but remember that if you turn off now when you re-turn on you reload them from eeprom).
M500 then says to store current memory values to eeprom.
When you execute some commands,eg M851, you change in memory values, hence you have to execute M500 if you want then to be reused after power-off.

I hope to make some light in you and not confused more than now. ;-)

syedrizvi0512, 2018-05-24:
Firstly GMagician and thinkyhead thank you SO much for explaining and helping out. I understood from both of you and thats really surprising since I'm not good at coding at all and I'm actually understanding this and hacking on my own...but slowly since I don't want the nozzle to press in the bed.

So I have been spending the past hour bringing the `#define Z_PROBE_OFFSET_FROM_EXTRUDER 0.25` to -2.50 and I have noticed some change in the offset from the bed but its not a whole lot. Maybe 0.25"? And so I'm wondering if I'm going anything wrong.
In the most recent run I had the `#define Z_PROBE_OFFSET_FROM_EXTRUDER -2.50` and when I did a G30 I got the following code at the serial monitor

`start
echo:Marlin 1.1.8

echo: Last Updated: 2017-12-25 12:00 | Author: (Michael Henke, flsun Kossel Mini)
echo:Compiled: May 23 2018
echo: Free Memory: 3339  PlannerBufferBytes: 1264
echo:V47 stored settings retrieved (902 bytes; crc 45381)
echo:  G21    ; Units in mm
echo:  M149 C ; Units in Celsius

echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X100.00 Y100.00 Z100.00 E90.00
echo:Maximum feedrates (units/s):
echo:  M203 X200.00 Y200.00 Z200.00 E200.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X4000 Y4000 Z4000 E4000
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P2500.00 R3000.00 T3000.00
echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z10.00 E5.00
echo:Auto Bed Leveling:
echo:  M420 S0
echo:Endstop adjustment:
echo:  M666 X0.00 Y0.00 Z0.00
echo:Delta settings: L<diagonal_rod> R<radius> H<height> S<segments_per_s> B<calibration radius> XYZ<tower angle corrections>
echo:  M665 L218.00 R101.00 H280.00 S160.00 B75.00 X0.00 Y0.00 Z0.00
echo:Material heatup parameters:
echo:  M145 S0 H180 B70 F0
echo:  M145 S1 H240 B100 F0
echo:PID settings:
echo:  M301 P24.77 I1.84 D83.61
echo:Z-Probe Offset (mm):
echo:  M851 Z-2.50
echo:Hardcoded Default Settings Loaded
ok
echo:SD card ok
echo:Settings Stored (902 bytes; crc 45381)
ok
echo:busy: processing
echo:busy: processing
X:0.00 Y:0.00 Z:280.00 E:0.00 Count X:47319 Y:47319 Z:47319
ok
echo:busy: processing
echo:busy: processing
Error:Probing failed
X:0.00 Y:0.00 Z:-5.00 E:0.00 Count X:18819 Y:18819 Z:18819
ok
`
I also did a G1 X0 Y0 Z-7 to try to make it go lower. The effector moves and than when I do a M114 I get  location X:0.00 Y:0.00 Z:0.00

GMagician, 2018-05-24:
Axes can't move above or below limits (?_MIN_POS, ?_MAX_POS ?=X,Y,Z etc) this is why you get 0 on Z.
Usually Z=0 is nozzle almost in contact with bed (the usual sheet of paper under nozzle).
Yoou have to adjust Z_PROBE_OFFSET value to be sure to reach this goal.

thinkyhead, 2018-05-24:
Sorry we haven't had more time to give support. Here's the cut-and-paste message that we usually post when we're very busy…

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. For best results getting help with configuration and troubleshooting, please use the following resources:
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.

boelle, 2019-02-20:
@thinkyhead maybe close this one?

closed 2019-03-17 by p3p

themistoklisv, 2020-04-13:
I have the same message "Probe failed". I am using an anycubic kossel plus and 2 days ago I uploade a marlin 2 firmware. Then at some point I initiated eeprom and from then on the printer will not auto calibrate. I have tried reflashing, resetting in pronteface but nothning. Can you help?

thinkyhead, 2020-04-27:
@themistoklisv — 
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).