MarlinFirmware/Marlin #10800: I cant achieve babystepping with 1.1.8

bentech4u, 2018-05-20:
HI,

i was into 1.1.x-bugfix series and i changed to 1.1.8 and baby stepping is not working for me,

i did all the changes to enable babystepping like 1.1.x-bugfix and i can able to see the z offset/babystepping screen when i long press/double press on rotary switch. But when i turn the POT, Z level is not changing. 

usually i am not using G29 on gcode, i always used to adjust my first layer using babystepping only

Configuration_adv.h
`#define BABYSTEPPING
#if ENABLED(BABYSTEPPING)
  //#define BABYSTEP_XY              // Also enable X/Y Babystepping. Not supported on DELTA!
  #define BABYSTEP_INVERT_Z false    // Change if Z babysteps should go the other way
  #define BABYSTEP_MULTIPLICATOR 8   // Babysteps are very small. Increase for faster motion.
  #define BABYSTEP_ZPROBE_OFFSET   // Enable to combine M851 and Babystepping
  #define DOUBLECLICK_FOR_Z_BABYSTEPPING // Double-click on the Status Screen for Z Babystepping.
  #define DOUBLECLICK_MAX_INTERVAL 1250 // Maximum interval between clicks, in milliseconds.
                                        // Note: Extra time may be added to mitigate controller latency.
  #define BABYSTEP_ZPROBE_GFX_OVERLAY // Enable graphical overlay on Z-offset editor
#endif`

Configuration.h

`#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

#define FIX_MOUNTED_PROBE

#define AUTO_BED_LEVELING_BILINEAR

#define Z_SAFE_HOMING

#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT (30)    // X point for Z homing when homing all axes (G28).
  #define Z_SAFE_HOMING_Y_POINT (10)    // Y point for Z homing when homing all axes (G28).

#endif`

bentech4u, 2018-05-21:
Anyone please help me ,

bentech4u, 2018-05-21:
i flashed below:

[https://github.com/ejtagle/Marlin/archive/bugfix-2.0.x.zip](url)

only Babystepping /z offset is not working while printing, apart from that i can screen/pass menus in normal operations

Bob-the-Kuhn, 2018-05-21:
Please try the latest bugfix-2.0.x

It was just updated yesterday to make babystepping work on the 32 bit CPUs.  The same fix was also moved into bugfix1.1.x.

bentech4u, 2018-05-21:
the reason for the update is #10446 . is this fixed on bugfix-1.1.x ?

Bob-the-Kuhn, 2018-05-21:
I don't think it's been fixed.  The last comment I see in there is that they're still seeing it sometimes.

bentech4u, 2018-05-21:
ok. seems the download link they put on that thread fixed my skipping issue. almost finishing one model which failed 3-4 times earlier

Bob-the-Kuhn, 2018-05-21:
Please post your results in that thread.

Can we close this issue?

bentech4u, 2018-05-22:
i am not sure, i will try more test prints

thinkyhead, 2018-05-22:
Note also that the double-click to access babystepping was also patched, so it can't be activated outside of printing. Since babystepping is only allowed (at present) while moves are underway.

thinkyhead, 2018-05-22:
So, what have you got there? A delta, CoreXY, cartesian…?

bentech4u, 2018-05-22:
homemade p3steel, kind of Prusa design
arduino mega+ ramps 1.5+ A4988 electronics.

thinkyhead, 2018-05-22:
A reversed direction is curious. Is that consistent? Up is always down and down is always up?

bentech4u, 2018-05-22:
yes, it was opposite initially, Then I enabled. Please don't consider this as confirmation as i am very new to marlin, and need to double check again 

`#define REVERSE_ENCODER_DIRECTION
#define REVERSE_MENU_DIRECTION`

pfjason, 2018-05-23:
I'm also seeing this issue on the bugfix branch at 5bb295ef6a71b948eed75c02d71540ebf3664553. I'm not able to double click for baby step.

pfjason, 2018-05-23:
Scratch that, doubleclick to babystep with Babystep_zprobe_offset turned off does work, but only while printing apparently. I'll test it the other way.

bentech4u, 2018-05-23:
@pfjason it activates only while printing i guess.

bentech4u, 2018-05-23:
Is there any issue on stable 1.1.8. Again i flashed 1.1.8 today and the same behavior. Baby stepping is activated while print starts and when i rotate the switch, values are changing on the screen. but z-axis not moving

pfjason, 2018-05-24:
Mine works as expected, except for it not really saving the offset when changed via babystepping as described above on stable 1.1.8. I haven't had a chance to test the bugfix with BABYSTEP_ZPROBE_OFFSET turned back on for that bug yet.

thinkyhead, 2018-05-24:
> Again i flashed 1.1.8 today and the same behavior.

1.1.8 will never change. So if things are broken in that version, you'll have to use another.

bentech4u, 2018-05-24:
i don't know true or not, 

i got baby stepping working on [https://github.com/ejtagle/Marlin/archive/bugfix-2.0.x.zip](url) this. but the print quality was very poor, i can see some lines on print. 

again flashed 1.1.8 , printed the same model. print was perfect,

is this expected or anyone reported yet?

bentech4u, 2018-05-24:
i will try to flash another 

[https://github.com/MarlinFirmware/Marlin/tree/2ca0b732a123ec5bc3384489b8129fb2784bbadb](url)

thinkyhead, 2018-05-24:
There are some unresolved issues with the current bugfix branches, as we've been overhauling the planner to improve performance. In some ways it is better, and in some ways it isn't. You may need to tune the acceleration and jerk differently from 1.1.8 to get equivalent results with the latest code. We hope to get the planner issues sorted out soon so that results are more consistent.

unixb0y, 2018-09-14:
Hey guys, in my case it's a bit different from anything else I've read yet:
* I have Marlin 1.1.8 
* Babystepping menu with double click does open when not printing
* But it doesn't do anything

My configuration:
```
#define BABYSTEPPING
#if ENABLED(BABYSTEPPING)
  //#define BABYSTEP_XY              
  #define BABYSTEP_INVERT_Z false   
  #define BABYSTEP_MULTIPLICATOR 10   
  #define BABYSTEP_ZPROBE_OFFSET   
  #define DOUBLECLICK_FOR_Z_BABYSTEPPING 
  #define DOUBLECLICK_MAX_INTERVAL 1250 
  //#define BABYSTEP_ZPROBE_GFX_OVERLAY 
#endif
```

ruggb, 2019-02-11:
Is this issue ever going to get resolved? It does not work in V2 either. I have tried V2 bf 870757a and v1.1.9 bf 3a216bb. This worked once long ago but got broken somewhere along the line.
Babystepping works ONLY during printing from the TUNE menu. Doubleclicking is not functional at any time.

boelle, 2019-02-20:
@bentech4u is this still an issue with latest bugfix 2.0?

ruggb, 2019-02-23:
other than the fact that it does not open if not printing - no it is not an issue.

closed 2019-02-23 by thinkyhead