MarlinFirmware/Marlin #10804: Homing Z axis fails with dual Z TMC2130 drivers

lucasltt, 2018-05-21:
Hi,

I spent most of my weekend trying to figure out exactly what is going on and provide a clear path here, but I'm afraid to go on the wrong direction so I'll describe my findings here meanwhile.

I have a dual Z TMC2130 drivers in my printer (also on X and Y axis). If I ask the printer to go Z+ or Z- it is obeying well. However, when homing Z, BLTouch Hits the bed but the axis keeps descending crashing into the bed.  I first thought of some issue with BLTouch, but when it is on the Alarm state, M119 reports Z as triggered.

After doing some compiling and testing, I found out that the endstop update condition (on endstops.cpp) was not being matched due to the "motor_direction" function returning false. I later found out that on _Steppers.cpp_, method _set_directions_, variable _count_direction[Z_AXIS]_ is positive when Axis is going down and negative when axis is going up. I made a workaround just test if homing would succeed.

`void Stepper::set_directions() {

  #define SET_STEP_DIR(A) \
    if (motor_direction(_AXIS(A))) { \
      A##_APPLY_DIR(INVERT_## A##_DIR, false); \
      count_direction[_AXIS(A)] = -1; \
    } \
    else { \
      A##_APPLY_DIR(!INVERT_## A##_DIR, false); \
      count_direction[_AXIS(A)] = 1; \
    }

  #if HAS_X_DIR
    SET_STEP_DIR(X); // A
  #endif
  #if HAS_Y_DIR
    SET_STEP_DIR(Y); // B
  #endif
  #if HAS_Z_DIR
    SET_STEP_DIR(Z); // C
    count_direction[Z_AXIS] = count_direction[Z_AXIS] * -1;
  #endif

  #if DISABLED(LIN_ADVANCE)
    if (motor_direction(E_AXIS)) {
      REV_E_DIR();
      count_direction[E_AXIS] = -1;
    }
    else {
      NORM_E_DIR();
      count_direction[E_AXIS] = 1;
    }
  #endif // !LIN_ADVANCE
}`

And it indeed homing does work. However I could not figure out why the function reports direction inverted.

Some side notes:
- Using 1.1.x bugfix branch
- When using DRV drivers, the issue does not exists.
- The same behavor was not seen on X and Y axis.

Thanks in advance

thinkyhead, 2018-05-25:
We've made several alterations to the endstop handling, so please try the latest code and see if it behaves better for you.

lucasltt, 2018-06-02:
For anyone running with the same issue. Seems that any microstep higher than 32 will produce a random behavior on the direction of the motors and endstop detection.  I

closed 2018-06-02 by lucasltt