MarlinFirmware/Marlin #10806: [1.1.x]  #error "You are using an old Configuration.h file, update it before building Marlin."

bruce356, 2018-05-21:
Hi Just downloaded bugfix 1.1.x tried to compile with Arduino v1.8.6 using both the default files as down loaded and with my previous configs transfered and get the following errors on compiling.
SanityCheck.h:47: error: #error "You are using an old Configuration.h file, update it before building Marlin."
SanityCheck.h:51: error: #error "You are using an old Configuration_adv.h file, update it before building Marlin."
Its probably something very simple to fix.
regards - bruce

iosonopersia, 2018-05-21:
> #error "You are using an old Configuration.h file, update it before building Marlin."

Have you even read the error itself? I think it's enough self-explanatory...

bruce356, 2018-05-21:
Hi iosonopersia, thanks for the quick response, I know I am not a programmer I just downloaded the latest zip file left the default settings compliled (or tried) and received the above error. Please tell me why I am stupid.

iosonopersia, 2018-05-21:
I never said you're stupid... Anyway, compiling with the default Configuration.h and Configuration_adv.h should not raise those errors. I quickly checked the code and everything is ok. Make sure that you downloaded the actual bugfix 1.1.x and that you didn't touch any of the files. Than, it may be a problem of Arduino IDE, I really don't know

bruce356, 2018-05-21:
I know you did not but I thought that I missed something simple that was my fault.
I just downloaded again touched nothing, just compiled with Arduino set to Mega 2560 and same errors as above.

Just to make sure I unzipped the file to my laptop that has Arduino v1.8.5 installed and it compiled normally NO Errors? What the f I don't get it. Thanks again for spending the time and helping.
Regards - bruce

closed 2018-05-21 by bruce356

siana, 2018-05-21:
You have copied Configuration.h and/or Configuration_adv.h from an older version of Marlin. This error message occurs when the settings in Configuration.h have changed in structure or in meaning, so the files you have substituted are incompatible, and you need to write a new set of configuration files.

The last changes that happened that would trigger the error happened in 1.1.7 and 1.1.0.

To write the new file, you take the configuration files that come with the latest version of Marlin, and for each setting in a new file version, check whether this setting is present in old configuration files, and manually copy it over to the new files; while doing so, mark all the settings in the old files that you have carried over. Then go through the settings that are left unmarked, i.e. the ones you haven't carried over, and think whether you've seen a new setting that roughly corresponds to them and you need to change something, or whether the setting was unimportant and can be left out.

Let us know if you hit a roadblock.

The way Marlin detects that the file is obsolete is by evaluating the #define CONFIGURATION_H_VERSION on top of Configuration.h and #define CONFIGURATION_ADV_H_VERSION on top of Configuration_adv.h; however if you were to cheat and just patch them up, you WILL run into more issues.