MarlinFirmware/Marlin #10815: Poor results with A8 Marlin config

bill-orange, 2018-05-22:
Using an Anet A8 with the stock extruder, the example configs give poor results

### Steps to Reproduce

compile and run Marlin with example configs on an Anet A8

**Expected behavior: calibration cube generally comparable to that generated with stock firmware

**Actual behavior: poor quality calibration cube (see photo)

#### Additional Information

[image]

ManuelMcLure, 2018-05-22:
That looks like there's some overextrusion. Try changing the last entry in `DEFAULT_AXIS_STEPS_PER_UNIT` to 98 (from 100) and see if that helps. I've been using essentially the A8 default configuration and get excellent results.

What version of Marlin are you running? 1.1.8 or bugfix-1.1.x? If the latter, what date?

bill-orange, 2018-05-22:
I am running 1.1.8.  This is what is so weird, nearly everyone is running the default successfully.  I was running the stock firmware successfully. There is nothing unique about my setup that I am aware of.  So why am I 'special' :)

ManuelMcLure, 2018-05-22:
The 98 instead of 100 recommendation is from personal experience. I was getting overextrusion with the default settings and that reduction helped a lot. I'm now back to 100 because I changed the extruder gear for a hobbed bolt but I needed that change to get good results with the factory extruder gear.

thinkyhead, 2018-05-22:
Also, please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. Don’t forget to issue `M502`, `M500` after flashing.

Squid116, 2018-05-22:
> That looks like there's some overextrusion. Try changing the last entry in DEFAULT_AXIS_STEPS_PER_UNIT to 98 (from 100) and see if that helps. I've been using essentially the A8 default configuration and get excellent results.

Rather than randomly changing your esteps, I think you'd be better off leaving it at your theoretical mechanical value (i.e. 100) and adjust your flow rate using a calibration routine.

This is my goto for calibration:  http://www.desiquintans.com/flowrate

Note: that because I use slic3r my Extrusion width is not the same as s3d, so I can't use his calculator, but the formula holds true.

bill-orange, 2018-05-22:
Turned out my extruder fan was failing.  After setting it for a higher voltage, I was able to test Marlin successfully.

So ...... false alarm.

closed 2018-05-22 by bill-orange