MarlinFirmware/Marlin #10816: FLSUN Kossel Big Size with Heated Bed Auto leveling issue.

ItsDrunkLuck, 2018-05-22:
I have just purchased a new 3D Printer and i cant get the thing working. I'm kind of new to all of this but i have a basic understanding of it. the original issue was that when i tried to auto calibrate it, it would just ram the nozzle into the bed. so i downloaded the debug version of the configuration for Marlin  1.1.8, and activated the "DEBUG_AUTO_LEVELING" or something like that. The problem now, is that when i try to run the code "G29" it doesn't move the Y tower motor at all.
Help?

ItsDrunkLuck, 2018-05-22:
@thinkyhead I saw a thread that was created around April 22nd that you posted on, and I followed through with the steps and its not really working.

ItsDrunkLuck, 2018-05-22:
@thinkyhead sorry im sending  so many, here is a video of what happens. the tower on the far end is my X, the one closest on the left is my Y, and the right is my Z. [https://drive.google.com/open?id=1K7EHEITUWO_LH-1kAv7TMGqdhtFzwkQk](url)

thinkyhead, 2018-05-23:
What does `M119` say when the carriages are not hitting the endstops? And then what does it say when you trigger each of the endstops? Do they all appear to be working correctly?

ItsDrunkLuck, 2018-05-23:
I've been messing with it a bit and i got rid of that problem, problem is, now its ramming the bed again. ill send another video. Ill also be sending a video of what happens when it homes on the X and Y, it does one tower at a time and i don't know if it is normal. G28: [https://drive.google.com/open?id=1jBMQ3uZIEtl_M_p1sCNsxHL8lL-XhDBN](url) G29: [https://drive.google.com/open?id=1b3ZSRbm_EyeBHFujfYIM9h0ehtfTwLb0](url)

ItsDrunkLuck, 2018-05-23:
@thinkyhead

ItsDrunkLuck, 2018-05-23:
i checked the endstops and they all seem to be working

ItsDrunkLuck, 2018-05-23:
well no, my Z max is always triggered regardless whether or not the switch is activated.

boelle, 2019-02-20:
@ItsDrunkLuck still having an issue if using latest bugfix 2.0?

if its not a marlin issue then maybe hit the green close button below

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead