MarlinFirmware/Marlin #10817: POWER_LOSS_RECOVERY format error

cdedwards, 2018-05-22:
The format in the PSTR() was set for a string with %s. job_recovery_info.leveling is a float. So, changed the %s to a %i and then converted the float into an integer. This is all occurring in the src/feature/power_loss_recovery.cpp file at line 131       

#if HAS_LEVELING
          // Restore leveling state before G92 sets Z
          // This ensures the steppers correspond to the native Z
          sprintf_P(job_recovery_commands[ind++], PSTR("M420 S%i Z%i"), int(job_recovery_info.leveling), int(job_recovery_info.fade);
        #endif

thinkyhead, 2018-05-23:
Thanks! We actually need the fade value as a `float`, so…
```cpp
char str_1[16], str_2[16];

#if HAS_LEVELING
  // Restore leveling state before G92 sets Z
  // This ensures the steppers correspond to the native Z
  dtostrf(job_recovery_info.fade, 1, 1, str_1);
  sprintf_P(job_recovery_commands[ind++], PSTR("M420 S%i Z%s"), int(job_recovery_info.leveling), str_1);
#endif
```

cdedwards, 2018-05-23:
ok. works for me.

commit 3cd5b95e (2018-05-23): Fix fade value for power-loss recovery

Fix #10817
commit 773a5504 (2018-05-23): Fix fade value for power-loss recovery

Fix #10817
closed 2018-05-23 by cdedwards