MarlinFirmware/Marlin #10824: Y axis reversed.

klausbfrederiksen, 2018-05-23:
I use Marlin 1.1.8 on an I3. 
I have realized that my Y axis is reversed:

My endstops are in Xmin, Ymax, Zmin, and X_HOME_DIR -1, Y_HOME_DIR 1 & Z_HOME_DIR -1.
And I use USE_XMIN_PLUG, USE_YMAX_PLUG, USE_ZMIN_PLUG.

When I home X, it homes correct to the left. Homing Y moves buildplate to the rear. Again correct.
Here is the problem now: Having homed X & Y axes, position values are X=0 and Y=0.
I did expect Y to home at Y=0 in front of the buildplate.

If I do G28 x100 Y100, the extruder centers on the buildplate.
If I then do a G28 x0 y0, the extruder returns to the left, and rear of the buildplate.

Somehow the home position is at the rear of Y axis, and not the front.

Changing the endstop position to the front, do not change that Y=0 is at the rear.

Any suggestions to this problem?

thinkyhead, 2018-05-24:
If you have a moving bed with the endstop switch mounted at the back then your `Y_HOME_DIR` should be -1. Since `Y_HOME_DIR` set to 1 moves the bed towards the back, your Y axis is set to move in the wrong direction. You must either flip the plug connecting the Y stepper motor to the board, or change your `INVERT_Y_DIR` setting.

klausbfrederiksen, 2018-05-24:
It actually works (had to connect endstop to Ymin [USE_YMIN_PLUG]- thank you for the solution.
I find it to be a bit reversed :-) here is why. I have used this figure, as a guide https://3dprinting.stackexchange.com/questions/2593/how-does-the-home-position-work-on-a-prusa-i3-reprap-printer
From this, the Ymax is to the rear of the buildplate, not to the front.
I do not find the solution logical at all. But it works, so I am a happy man.
Stay Cool - I try to in 25 C

thinkyhead, 2018-05-24:
> had to connect endstop to Ymin

If the endstop is at the back of the machine, that is correct.
If you had an XY moving gantry, the Ymin endstop would be in front.

In both cases, Y=0 has the nozzle aligned to the front edge of the bed.

sidhant, 2019-01-02:
@thinkyhead I reached this thread because I have the same confusion. Please help?

My config:
```
#define USE_XMIN_PLUG
//#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG
//#define USE_XMAX_PLUG
#define USE_YMAX_PLUG
//#define USE_ZMAX_PLUG

// Invert the stepper direction. Change (or reverse the motor connector) if an axis goes the wrong way.
#define INVERT_X_DIR false
#define INVERT_Y_DIR true
#define INVERT_Z_DIR false

// Direction of endstops when homing; 1=MAX, -1=MIN
// :[-1,1]
#define X_HOME_DIR -1
#define Y_HOME_DIR 1
#define Z_HOME_DIR -1

// Invert the stepper direction. Change (or reverse the motor connector) if an axis goes the wrong way.
#define INVERT_X_DIR false
#define INVERT_Y_DIR true
#define INVERT_Z_DIR false

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0.5 //Disable software endstop beore moving to Z0 to allow this to happen
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 100

//#define MANUAL_X_HOME_POS 0
//#define MANUAL_Y_HOME_POS 0
//#define MANUAL_Z_HOME_POS 0
```

With this current config, when I home using Pronterface, I end up with the nozzle at X=0, Y = 120, Z=0.5.

```
SENDING:M503
echo:  G21    ; (mm)
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X93.00 Y93.00 Z400.00 E97.00
echo:Maximum feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z5.00 E25.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X3000 Y3000 Z100 E10000
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced: Q<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 Q20000 S0.00 T0.00 X10.00 Y10.00 Z0.30 E5.00
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Unified Bed Leveling:
echo:  M420 S0 Z10.00
Unified Bed Leveling System v1.01 inactive.
Active Mesh Slot: -1
EEPROM can hold 7 meshes.
echo:PID settings:
echo:  M301 P22.20 I1.08 D114.00
echo:Z-Probe Offset (mm):
echo:  M851 Z0.70
echo:Stepper driver current:
echo:  M906 X280 Y350 Z1000
  M906 T0 E1200
>>> G28
SENDING:G28
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
>>> M114
SENDING:M114
**X:0.00 Y:120.00 Z:0.50 E:0.00 Count X:0 Y:11160 Z:200**
>>> G28 X0,Y0
**SENDING:G28 X0,Y0**
echo:busy: processing
>>> M114
SENDING:M114
**X:0.00 Y:120.00 Z:5.00 E:0.00 Count X:0 Y:11160 Z:2000**
```

My Y-endstop is at the front of the Y-table, that is, if I push the table away from me all the way, it gets triggered. The endstop is plugged into the Y_MAX_PIN.

If I jog manually, the motor moves as expected. +Y move the table away from me (the way I think, the nozzle sees an increasing Y coordinate space below it). 

Why is my home position 0,120 with nozzle at left, front corner of bed. Executing a G1 F200 Y0, brings the bed all the way forward. With G1 F200 Y120, it pushes bed all the way back and engages the endstop switch.

Using Marlin-bugfix-1.1.x

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-03-14 by thinkyhead