MarlinFirmware/Marlin #10825: E Maximum Acceleration not working

Phaelz, 2018-05-23:
Hi all.

I am playing with the maximum settings on my printer and I've come to realize that E Maximum Acceleration and E Jerk are not working as I expected and as the other axis.

I am using the M code M201, that modifies Maximum Acceleration on each axis. When I modify the values on X, Y and Z the speed is limited by the value. But not for E.

Then I used M204, that modifies Acceleration on moves (Print, Retract and Travel), and then the E speed is limited. Only by M204, but not by M201.

Is it correct?

thinkyhead, 2018-05-24:
The factor set for E by `M201` does get applied in the planner along with the other axes. You could try adding debugging messages to the planner to see what's going on in your case.

`M204` should have a more visible effect, since it's the value that acceleration starts out with for E-only moves. The value for `M201 E` will probably only be evident if it is set very low, since E tends to move at a much slower rate than other axes do.

boelle, 2019-02-20:
@Phaelz did you try what thinky suggested?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead