MarlinFirmware/Marlin #10828: Einsy Rambo - Reprap Discount Full Graphics Garbled

Sloppsta, 2018-05-23:
### Description
I tried adding
#define ST7920_DELAY_1 DELAY_NS(0)
#define ST7920_DELAY_2 DELAY_NS(4)
#define ST7920_DELAY_3 DELAY_NS(0)
to the ultralcd...st2930.h and also in the config.h but I keep getting a garbled screen. I have tried mutiple values and none seem to help. Suggestions?

### Steps to Reproduce

1. hook up einsy rambo and reprap full graphics display
2. turn on.

**Expected behavior:** [What you expect to happen]
Screen not garbled
**Actual behavior:** [What actually happens]
Screen Garbled
[Nicks Marlin 1.1.9 v1 Einsy Rambo June 2018.zip](https://github.com/MarlinFirmware/Marlin/files/2032483/Nicks.Marlin.1.1.9.v1.Einsy.Rambo.June.2018.zip)

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

Sloppsta, 2018-05-23:
Thanks for the quick reply! I had no idea what a nope even was :P in the board definitions they are only adjusting the middle value (delay 2) should all three be adjusted or just keep trying to up delay 2? Thanks!

ejtagle, 2018-05-23:
Try setting them all to 900, then reducing one by one. Those delays start adding quite a bit of time to the display handling, so the less the delays, the better

Sloppsta, 2018-05-23:
I've worked my way up to `#define SDSUPPORT
	#define ST7920_DELAY_1 DELAY_NS(100000)
	#define ST7920_DELAY_2 DELAY_NS(100000)
	#define ST7920_DELAY_3 DELAY_NS(100000)`

With no notable improvement. Any suggestions?

amoose136, 2018-05-24:
I have the EinsyRambo and a RepRapDiscount full graphics display but have not yet hooked the two together. I'll report back if I encounter any problems/solutions to problems.

larsnordstrom, 2018-05-24:
Enable #define DOGLCD

thinkyhead, 2018-05-24:
Be sure to check that the keyed plugs aren't backwards. Sometimes this happens on boards, sometimes on displays. On the Einsy the key slots should be facing inwards.

Sloppsta, 2018-05-24:
Thank you all for replying. @thinkyhead , I am a particular fan of your work!

I read somewhere else that doglcd is enabled automatically, so I didn't think I needed to do that.

Strangest thing though,after trying to set my delays to 900ns each as @ejtagle suggested, I decided to call it a miniRambo, and copied over the einsy pins to pins_minirambo- it did not work. When I flashed it back to einsy with the same config - it was working perfectly! All subsequent flashes have worked great too! I have no idea what happened but I am glad it's working great!

closed 2018-05-24 by Sloppsta

thinkyhead, 2018-05-25:
@Sloppsta — Good news! And thanks for the kind words. Will you still be a fan if my ERRF2018 keynote crashes and burns? I'm so not ready!

Sloppsta, 2018-05-25:
It will be awesome! Wish I could go!