MarlinFirmware/Marlin #10830: Marlin bubfix 1.1.x No more access to the printer menu (Creality CR-10)

Hebign, 2018-05-23:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

viper93458, 2018-05-23:
I had the same issue yesterday and downloaded the bugfix from today and the issue went away.

William

Hebign, 2018-05-24:
Thank you William for the info.. I'll try again

thinkyhead, 2018-05-24:
It was a typo. Should be fixed. Cheers!

closed 2018-05-24 by thinkyhead

Hebign, 2018-05-24:
Yes, tested today, fixed ! Thanks