MarlinFirmware/Marlin #10834: [1.1] Extruder auto fan is slow/under powered?

forkoz, 2018-05-24:
### Description
I have configured extruder auto fan on HE1 like so:

#define E0_AUTO_FAN_PIN MOSFET_D_PIN
#define EXTRUDER_AUTO_FAN_TEMPERATURE 50
#define EXTRUDER_AUTO_FAN_SPEED   255  // == full speed

The fan spins but seemingly too slow, I bought extra fast fan and its working just like my old fan.
I checked voltage from the heater pin and its 12.x volts

### Steps to Reproduce

Wire fan to DC power supply.
Wire fan to MKS GenL and use auto fan.

**Expected behavior:** [What you expect to happen]
Equal speed.

**Actual behavior:** [What actually happens]

Marlin fan is slow/quiet, 12v supply fan is faaaaast.

MikeLud, 2018-05-25:
Forkoz,

I have the same board and have the extruder fan setup the same way and it is working fine. Maybe you have some the wire insulation in the terminal and you are not making a good connection

Mike.

forkoz, 2018-05-25:
I will check.. I also found 12v power supply was putting out extra volts, like 15-16. That could account for it. Just because it says 12v on the box doesn't mean its 12v :(

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-03-13 by thinkyhead