MarlinFirmware/Marlin #10835: Tuning Z-offset on LCD Screen

Fourmi, 2018-05-24:
Hi,

Possible to make a feature for tuning Z offset on LCD screen ?
i make a google sheet to explain.
[Z-offset Tuning ](https://docs.google.com/spreadsheets/d/1qhyDCcUXyRs7eXB9-S-V7gBr81MqOKYM9TACI3_DAhE/edit#gid=0)

First step already run on my lcd screen, it's for the second .......

thx

thinkyhead, 2018-05-24:
It's on our TODO list already. Except the procedure would be:
1. Home XYZ, move XY to bed center.
2. User adjusts nozzle until it touches the bed (i.e., paper test)
3. A click sets this as Z=0, then a probe is done.
4. Z when the probe triggers is negated and saved as the Z offset.

Fourmi, 2018-05-24:
Ok, Nice Thinkyhead ;)

thinkyhead, 2018-05-25:
I hope it will work well! Setting it manually is such a hassle.

boelle, 2019-03-13:
@Fourmi did it work?

Fourmi, 2019-03-13:
Hi @boelle  !
Yes we make a menu on LCD for "touch_MI" sensor , but we don't make  pull request again, i think we can soon .
You can found the firmware on my github here >https://github.com/Fourmi/Marlin_TouchMI
Screenshot of menu >https://gyazo.com/02540dbb34fd0e98b62b86f86808fd57

boelle, 2019-03-13:
could you submit a PR to bugfix 2.0?

Fourmi, 2019-03-13:
Hum PR  on 1.1.9, i didn"t make on 2.0 but it's possible, but you can do if you want ;)

boelle, 2019-03-13:
1.1.9 is outdated hence i suggest 2.0.x

no i cant do it, i'm not a code person

InsanityAutomation, 2019-10-15:
@boelle @shitcreek I believe this is already done as well, and overlaps somewhat with babystepping

boelle, 2019-10-15:
we can always open it again if its not the case :-D

closed 2019-10-15 by boelle