MarlinFirmware/Marlin #10837: [Question/FR] z_offset on switch homed Z axis

GMagician, 2018-05-24:
My system homes Z using 2 switches and I need them because one stepper (due to head weight) is moving when printer is turned off, hence I use the 2 switches to be sure to have Z planar with bed (offsetting with Z_DUAL_ENDSTOPS_ADJUSTMENT).
Home position is set by MANUAL_Z_HOME_POS.
My personal case is special because once homed nozzle is 3mm above the bed (I use switches like a probe) and also I have a probe to let me used UBL.

The problem I have is when nozzle is changed, in this situation nozzle position may be higher/lower.
If I'm not missing something to adjust new Z position the only possibility is to recalibrate MANUAL_Z_HOME_POS, re-upload firmware and use printer.

As I told before my homing is more like a probe homing so z_offset could be effective but at the moment is not.

May be a change in such logic be useful also to other people?

thinkyhead, 2018-05-24:
> once homed nozzle is 3mm above the bed

So if your nozzle ever goes below the switches, do they stay triggered? And, I assume you're having Z raise before homing…

To have the Z home adjust, you can use `M206 Z`. But you have to make sure your Z_MIN_POS is set below the bed so that you have some extra freedom of movement to make adjustments.

GMagician, 2018-05-24:
Yes to both questions.
My switches will be pressed all the time when Z < MANUAL_Z_HOME_POS (that is 3.2), and I raise by 10mm before homing
Z_MIN_POS is 0 because this is Z value when nozzle touch the bed.
I'll see what M206 does to see if it act as I need

thinkyhead, 2018-05-24:
`Z_MIN_POS` can be a negative number, while `Z_HOME_POS` is 0. It's the `Z_HOME_POS` that matters with regard to the nozzle touching the bed when Z=0. `Z_MIN_POS` simply determines the physical limit and the software endstops.

GMagician, 2018-05-24:
If I'm not wrong by default Marlin does `#define Z_HOME_POS Z_MIN_POS` (negative direction homing), to have both Z_MIN_POS detached from Z_HOME_POS I have to define MANUAL_Z_HOME_POS.
Since I prefer z=0 (mental limit?) when nozzle touch the bed here comes `#define MANUAL_Z_HOME_POS 3.2`
Otherwise I had to define Z_MIN_POS to -3.2 and MANUAL_Z_HOME_POS to 0.

I don't know if M206 saves value into eeprom, if not it must be executed after every G28 (or after every turn on). But what happens if you load a .gode file? you have to add the proper M206 to each file.
Better is that z=0 (related to the nozzle) is always on the same mechanical position.
Z_OFFSET acts this way with the probe, it can be changed and store.

GMagician, 2018-05-24:
```c++
#define HAS_POSITION_SHIFT DISABLED(NO_WORKSPACE_OFFSETS)
#define HAS_HOME_OFFSET (DISABLED(NO_WORKSPACE_OFFSETS) && DISABLED(DELTA))
#define HAS_WORKSPACE_OFFSET (HAS_POSITION_SHIFT || HAS_HOME_OFFSET)
```
it seems that HAS_WORKSPACE_OFFSET and HAS_POSITION_SHIFT are the same thing!

GMagician, 2018-05-24:
@thinkyhead I'm trying to figure out how/where M206 works but I'm in trouble to follow flow.

In aonther part of the code there are two different test that are the same

```c++
#if HAS_HOME_OFFSET
#if HAS_HOME_OFFSET && HAS_POSITION_SHIFT
```
is just HAS_HOME_OFFSET.
It's a little bit complicated to understand, lol, I don't really know how you guys walk into this code with such  agility.

thinkyhead, 2018-05-25:
There has been some flux in the handling of workspace offsets. There may still be some legacy oddities.

GMagician, 2018-05-25:
@thinkyhead  may I try to cleanup an post a PR?

closed 2018-05-25 by GMagician