MarlinFirmware/Marlin #10838: Filament feed rate

rsmith21, 2018-05-24:
### Description

Filament feed rate
### Steps to Reproduce

1. [print in progress]
2. [extruder  starts to click]
3. [? I wonder if its a jam, or is the stepper trying to feed more than my system can work at]

**Expected behavior:** [Look at the display]

**Actual behavior:** [Either the feed rate is within my systems ability, or I have a blockage, or I have to slow down the Flow rate]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

thinkyhead, 2018-05-24:
We're quite busy tracking bugs, but we'll try to implement this in the 2.0 branch before the end of 2018.

dorfman2, 2018-11-20:
I support this feature as well. Thank you for listening!