MarlinFirmware/Marlin #10840: Question re UBL and upgrading firmware

modem7, 2018-05-24:
Hey guys, 

I'm going to be upgrading from 1.1.4 to 1.1.8, and was wondering how to port across my UBL. 

Is it just a case where I select the mesh from EEPROM? Or do I have to redo everything again? 

Many thanks!

ManuelMcLure, 2018-05-24:
One way of saving and restoring the mesh is to run `G29 S-1` - this will print out a bunch of `M421` commands to recreate the mesh. Copy and paste them to a local file, and then replay them when you've upgraded the firmware. It's a good idea to turn off temperature auto-reporting with `M155` before doing this to avoid getting temperature messages in the middle of your mesh data:
```
M155 S0
G29 S-1
M155 S5
```

I'm not sure if the mesh format has changed between 1.1.4 and 1.1.8, if it hasn't you may not need to do this (the mesh is stored at the end of EEPROM so it shouldn't be affected by a firmware upgrade) but it's a good idea to save the mesh anyway just in case.

modem7, 2018-05-24:
Thank you very much!

closed 2018-05-25 by thinkyhead