MarlinFirmware/Marlin #10842: FLSUN Delta G33 Escalates Quickly

Floppynator, 2018-05-24:
Hi Guys,

you are my last hope. i have a Flsun Delta and i dont get G33 runs well. No matter wich Firmware i use. In this case i used 1.1.x-bugfix branch.

I think a video says more than 1000 word.. look here. 
G28 Works well, also G1 Z10 F5000 for example
https://youtu.be/vPvUQfIGAH0 

Here you can see what happened i run G33 after G28
https://youtu.be/yyxTxGpoCMQ

any ideas whats happened? i tryed another driver but have the same issue. also with Marlin 1.1.8

please help me :(

Floppynator, 2018-05-24:
AUTO_BED_LEVELING_UBL and AUTO_BED_LEVELING_BILINEAR also do the same :(

Floppynator, 2018-05-24:
oh wow, now i get it. 
THe Spring from z-enstop probe are loose and there triggered random z-min...

holy moly

Floppynator, 2018-05-25:
Shure. But it would be Nice the Printer Not run over the printable Max and stop his movement :) Look at the Second Video, i Hit the panic Button. Otherwise the effector brake off :)

thinkyhead, 2018-05-25:
@LVD-AC — What do you make of the second video?

Floppynator, 2018-05-25:
G33 Delta Autocalibration