MarlinFirmware/Marlin #10845: [FR] Screensaver / idle-timeout on LCD

bipsendk, 2018-05-25:
### Description

Add functionality to disable/blank display (and maybe define a control pin or similar to switch on/off display backlight). 

If a display is idle for more than a defined number of minutes, the display will be blanked (and maybe backlight powered off). When the display/panel is operated again, the display should turn back on (and enable backlight if switched off), and "idle-wait-timer" start again to see if the display should be "powered off".
Whenever the printer is reset, the idle-wait-timer should be reset, so the timeout always will be calculated from the last operation on the display/control panel.

It should be possible to define whether the display is allowed to go into "sleep mode" if the printer is printing (maybe through a custom command to turn display on/off). Otherwise the display should only switch on again if the panel is operated - or when the print job has finished.

This functionality would be nice in order to avoid burn-in on some types of displays.

thinkyhead, 2018-05-25:
Duplicate of #4728

closed 2018-05-25 by thinkyhead

bipsendk, 2018-05-26:
Just for your information - the SSD1309 chip has a command to turn the display off (usefull with OLED displays). The functionality could be implemented to only support displays, where the controller has this sort of capability.

thinkyhead, 2018-05-27:
@bipsendk — Good to know. We can not only save on computation by not sending data, we can save energy too!

bipsendk, 2018-05-27:
The datasheet: http://www.allshore.com/pdf/ssd1309.pdf

section 10.12 Set display ON/OFF

Another useful function: Invert display (function described at in section 10.10)

Not sure whether the libraries for Arduino supports these functions - but it might be worth investigating...

thinkyhead, 2018-05-27:
> The amount of energy saved is minimal compared the the consumption of the rest of the system (especially when using a heated bed) ;-)

It's true, but if we save up enough electrons we can redeem them for valuable prizes!