MarlinFirmware/Marlin #10846: Extruder Overheated Marlin 2.0.x

pecirep, 2018-05-25:
### Description

I started a gcode from my SD-Card and left the room, upon my return a few minutes later I could smell my esun abs+ burning and quickly unplugged and replugged my printer, before even looking at the screen to see what's happening. After reboot the hotend was at 309C (instead of 230).

I printed close to a hundred prints with S3D-generated gcode on the same settings, half of those with Marlin 2.0.x, the config is posted below.

### Steps to Reproduce

I don't really know. I have restarted printing to see if the gcode was bad, but it's printing fine. Thermistor doesn't seem to be at fault either because it's correctly reporting temperature. I have not physically checked the thermistor as of yet and I would prefer not to unless someone can explain why it only temporarily stopped working correctly.

In case it matters, My last reboot of the machine was about 12 hours ago and I've done 2 (very similar) prints since and left the machine powered on but doing nothing for a couple of hours after the second print finished. The overheating happened on the first print after this long "pause". Again, don't know if that matters but it's the only thing I've done differently from usual.

#### Additional Information
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2040288/Marlin.zip)

ManuelMcLure, 2018-05-25:
Just to eliminate a possibility, check the MOSFETs on your RAMPS board (I'm assuming you're using RAMPS since you have `MOTHERBOARD` set to `BOARD_RAMPS_14_EFB` and make sure the metal tabs aren't touching each other. This can cause strange heating issues.

thinkyhead, 2018-05-25:
If Marlin was running and received a reading from your thermistor that the temperature was way out then it would have shut down the machine within 40 seconds, given your current settings. If Marlin was running and had been unable to get a temperature reading for any length of time (>=4 seconds), then the watchdog would have reset the board.

It sounds like you may have had a catastrophic lock-up that seized the firmware and left the heaters enabled. If the board had simply reset, then the heaters would have been disabled.

If you see this issue again, check to see if the controller is responsive. If it is not then there is some kind of seizure.

pecirep, 2018-05-26:
checked my mosfets, I've had successful prints since without changing anything and I am unable to reproduce the issue. If it happens again I'll check the controller as you said and reopen this issue.

closed 2018-05-26 by pecirep