MarlinFirmware/Marlin #10848: [FR] New GCODE for allowing slicer to tell Marlin the layer

marcio-ao, 2018-05-25:
We would like to implement a new GCODE to allow the slicer to tell Marlin which layer is being printed. This is primarily to be used by our backlash compensation code (#10826), to turn off the feature after a certain number of layers, but some users have also requested this information in the LCD, so that could be somewhere else where having this info available to Marlin would be useful.

It seems like in general having Marlin being layer aware would be a good thing. I can see this being a new GCODE, or perhaps a new argument to G92 (how about G29 Ln)? Thoughts?

thinkyhead, 2018-05-25:
I don't think it makes sense to overload `G92` for this. Instead, it would be better to adopt [M532 L](http://reprap.org/wiki/G-code#M532:_Set_print_progress) (making it a partial alias of `M73`).

marcio-ao, 2018-08-20:
We no longer need this feature as we found another way to accomplish the same thing. Closing this FR.

closed 2018-08-20 by marcio-ao