MarlinFirmware/Marlin #10850: Auto Calibrate Delta - Probe Failed.

ItsDrunkLuck, 2018-05-26:
Every time i use G33, it gets through the probing and ends up failing.

22:27:42.897 : echo:endstops hit:  Z:4.87
22:27:53.675 : echo:Unknown command: "G29"
22:27:58.623 : G33 Auto Calibrate
22:27:58.625 : Checking... AC
22:27:58.628 : .Height:300.00  Ex:+0.00  Ey:+0.00  Ez:+0.00  Radius:101.00
22:27:58.630 : .               Tx:+0.00  Ty:+0.00  Tz:+0.00
22:28:03.622 : echo:busy: processing
22:28:08.622 : echo:busy: processing
22:28:13.657 : echo:busy: processing
22:28:18.656 : echo:busy: processing
22:28:21.832 : Error:Probing failed
22:28:21.834 : Correct delta settings with M665 and M666

thinkyhead, 2018-05-26:
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as requested in the Issue template) and drop them into your next reply. We'll check them over and see if anything is amiss.

thinkyhead, 2018-05-27:
Is your "Z" max endstop different from the other two?
```cpp
#define X_MAX_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
#define Y_MAX_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
#define Z_MAX_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
```

ItsDrunkLuck, 2018-05-28:
@thinkyhead did you see any issues?

thinkyhead, 2018-05-28:
Nope, I didn't see any other issues. Last we knew `G33` was working. Are you able to do successful probes with `G29` and `G30` ?

ItsDrunkLuck, 2018-05-28:
yeah, no. i cant get a successful probe in.

ItsDrunkLuck, 2018-05-28:
im going to try a couple more times.

thinkyhead, 2018-05-28:
Let's gather more information…

- Download Marlin `bugfix-1.1.x` or `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Perform a `G28` to do your standard homing procedure.
- Issue the command `M111 S247` to enable maximum logging.
- Do a `G29` to try and probe the bed.
- Copy the log output into a **.TXT file** and attach it to your next reply.

ItsDrunkLuck, 2018-05-28:
[Log.txt](https://github.com/MarlinFirmware/Marlin/files/2044809/Log.txt)

cdedwards, 2018-05-28:
Try setting your delta height to a value much greater than the Z you think you have. M665 Hxxx.x
Then G33 P1  to calculate the height.

thinkyhead, 2018-05-29:
@ItsDrunkLuck — Your log has no details. Are you certain you used `M111 S247` after `G28` and before `G29`? If you did do the `M111` command, watch out for Repetier Host undoing it with its own debug buttons. You may be better off using the Serial Monitor from within Arduino IDE, which doesn’t mess with debug levels or filter the output.

boelle, 2019-02-20:
@ItsDrunkLuck is this still an issue?

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-03-13 by thinkyhead