MarlinFirmware/Marlin #10856: [Discussion] Config names and units for JUNCTION_DEVIATION and BEZIER_JERK_CONTROL

Sebastianv650, 2018-05-26:
Playing around with our new features, I realized we can maybe improve the content around the config files for them.

This is how the section for `JUNCTION_DEVIATION `inside Configuration_adv.h looks at the moment:
```
//
// Use Junction Deviation instead of traditional Jerk Limiting
//
//#define JUNCTION_DEVIATION
#if ENABLED(JUNCTION_DEVIATION)
  #define JUNCTION_DEVIATION_FACTOR 0.02
  #define JUNCTION_ACCELERATION_FACTOR 1000
  //#define JUNCTION_DEVIATION_INCLUDE_E
#endif
```

Maybe it's only me, but the word "factor" sounds like a magic value without a unit to me. I guess the average user will not know why we have two values to tune junction speeds and what they mean. Giving them a unit and a short description might help.
Here is a proposal, feel free to comment:

```
//
// Use Junction Deviation instead of traditional Jerk Limiting
// Read this article for details: https://onehossshay.wordpress.com/2011/09/24/improving_grbl_cornering_algorithm/
//
//#define JUNCTION_DEVIATION
#if ENABLED(JUNCTION_DEVIATION)
  #define JUNCTION_DEVIATION_FACTOR 0.02       // Distance from real junction edge in mm
  #define JUNCTION_ACCELERATION_FACTOR 1000    // Max. allowed centripetal acceleration in mm/s²
  //#define JUNCTION_DEVIATION_INCLUDE_E
#endif
```

Maybe even renaming `JUNCTION_DEVIATION_FACTOR `to `JUNCTION_DEVIATION_MM `and `JUNCTION_ACCELERATION `or `DEFAULT_ACCELERATION_MM_S2`.

Regarding `BEZIER_JERK_CONTROL`, don't you think this name will confuse users as it's called a jerk control but it is a replacement for the acceleration handling (= `BEZIER_ACCELERATION_CONTROL`)? While it is designed to reduce jerk during acceleration, we used this term inside Marlin only to talk about the junction jerk up to now.

Inviting: @thinkyhead @ejtagle @hoffbaked, please extend if I missed someone.

thinkyhead, 2018-05-26:
> `BEZIER_JERK_CONTROL`

I agree with the idea of getting these names better now rather than later, especially this one. I'm not entirely allergic to the use of "jerk" to talk about the ringing that comes from sudden changes in acceleration, and this option does reduce that effect. But, to be more direct, it should probably use the more common term (according to Google), `S_CURVE_ACCELERATION`.

ejtagle, 2018-05-26:
It´s a little better than S-Curves, S-Curves only warranty acceleration to be 0 at start and end of movements. This is a little better. The bezier jerk controlled curve allows also to warranty derivative of acceleration (called jerk) is also 0 at the start and end of movement.
But, i am not opposed. It is just a matter of finding what people could understand

iosonopersia, 2018-05-26:
Well, then it could be called `BEZIER_CURVE_ACCELERATION`, this really is self-explanatory

thinkyhead, 2018-05-26:
I love Pierre Bézier and his clever method for describing curves, but I'm not sure the average 3D printer enthusiast should be bothered with his funny name. Also, a casual search of YouTube for "S-curve Acceleration" turns up more pertinent content than searching "Bezier Curve Acceleration." Unless we plan to make a distinction between S-curves and Bézier curves, I think "S-Curve" is a reasonable moniker for this option.

Sebastianv650, 2018-05-26:
That looks much better to me :)

teemuatlut, 2018-05-26:
+1 for S curves because Trinamic demoed it with a beer glass :beer:.

hoffbaked, 2018-05-26:
I like it. I was struggling to find a name better than JUNCTION_ACCELERATION, since it’s the cornering force that it’s deccelerating to achieve and not accually accelerating anything. But the units seemed all wrong to call it something like JUNCTION_CORNERING_FORCE.

ejtagle, 2018-05-26:
BTW: Let´s remove the "32bit MPU required...." It´s not true anymore 👍

closed 2018-05-26 by thinkyhead