MarlinFirmware/Marlin #10861: Z Homing for PCB milling

salihonur, 2018-05-26:
Hello,

I wanna make PCB milling. I have tried many CNC firmwares but no one was stabil. So I decided to modify my working Marlin 3D firmware.

I use PCB copper surface as end stop. But this is not min or max end. This is reference point. What I need.
1) When engraving bit touches the surface the point will be set as reference point.
2) Then after homing Z axis, Z axis should move up abit (like -0.2 mm) to avoid from scratch while X and Y homing.
3) Then Z axis should move to required value (for example +0,2mm to mill or +1,5mm to cut the pcb) .

So touch point should be reference point and Z axis can be move both up and down direction.

NOTE: The picture is an protoype. In the picture PCB is drilled pcb. But in real action it will be solid copper. Distances will be real distance not header pins.

Can you help how can I do? Thanks...

Picture: [https://pasteboard.co/HmYZXXr.jpg](https://pasteboard.co/HmYZXXr.jpg)

salihonur, 2018-05-26:
I found how to set the touch point as the reference point by "#define MANUAL_Z_HOME_POS 0". Now I need to know that how I can move Z a bit mm up after Z homing and before XY homing.

GMagician, 2018-05-26:
Have you tryed to use input like it's for a probe instead on min?

salihonur, 2018-05-26:
The probe setting is for bed leveling, isn't it? Can I use it for reference point?

Meanwhile my Z settings like Z_MIN_POS -22,   Z_MAX_POS 2

NeoCortex3, 2018-05-26:
You might wanna look at G38.3 and G92.
But as of Marlin 1.1.8 G38 is broken. I'm currently working on that.

DerAndere1, 2021-12-09:
Have you tried to change this ?https://github.com/MarlinFirmware/Marlin/blob/c3af6bd8ce731737d0812a0e554e37dc0779127f/Marlin/Configuration_adv.h#L836